ufrisk/pcileech-fpga #84: Screamer M.2 only compatible with specific PCIe slots

dank, 2021-01-19:
The Screamer device works great on Linux (target), no matter what PCIe or M.2 slot I have the device in.

However, when I switch over to Windows on the target machine, all slots fail to boot except the first PCIe slot. Often times I'm able to reach the lock screen and Windows is completely bricked at that point, other times I just get a black screen after the Windows boot spinner.

Interestingly, I'm still able to probe the memory from the bricked state. When I have the device plugged into the working slots, all functions work as expected.

This is normally not a problem because I can switch PCIe devices around but, I have a particularly large GPU in this PC. 

Are there some configuration steps I missed? From reading other issues on this repo, it seems like the Screamer is a big pain and I should've bought the Xilinix dev board instead.

ufrisk, 2021-01-20:
Is the Linux PC and the Windows PC the same PC?
Are you running the default FPGA bitstream that I provide, or do you have a custom one?

dank, 2021-01-20:
Yes, I’m dual booting on separate drives. I’m also using the default bitstream. I’ve tried compiling from source as well, but to no avail.

ufrisk, 2021-01-26:
apologies; but this is just super hard for me to replicate without having the hardware causing this. Other have had success in upgrading their UEFI/BIOS; disabling VT-d and NX-bit in UEFI/BIOS and so on. I don't believe this is your issue, but I've heard stranger things.

Otherwise I'm just out of ideas.

dank, 2021-05-31:
I’m not sure how I didn’t think of this before, but I believe this is due to running out of PCI lanes on my motherboard, not specific to any slots or order my devices are plugged in. 

The way I tested this before was flawed. I swapped my GPU with the screamer (on a PCI riser) and it worked, then I had no space left to plug my GPU back in, so I assumed it was related to the slot. 

I also had a second M.2 device, my nvme using up PCI lanes the whole time, which I never considered was an issue. 

I’m still not sure why it worked on Linux and not Windows. With all that in mind, any configuration of PCI/M.2 works as long as I respect that I only have a limited amount of PCI lanes for my devices.

closed 2021-05-31 by dank