ufrisk/pcileech-fpga #85: Raptor dma build guide

latsarbomba, 2021-01-22:
Good evening Ulf! Amazing project you have here, I noticed in the code base there are flags for the raptor dma board, does the build for that firmware follow the same steps as the pci screamer since it uses the same chip? 

Thank you for all the work you've done as well  I don't have much but hopefully I can buy you a coffee with a sponsorship

ufrisk, 2021-01-22:
Hi and super thanks for sponsoring, it's very nice since Github will sponsor me with an equal amount effectively doubling your sponsorship. Thank You 💖 

Yes, RaptorDMA is 100% compatible with the ScreamerM2. You may use the bitstream or sources for the ScreamerM2 to customize your RaptorDMA firmware. The only thing you may wish to change is the device id from 4 to 6. https://github.com/ufrisk/pcileech-fpga/blob/e498483489cdef4e61f9b718297593057ac69f45/ScreamerM2/src/pcileech_screamer_m2_top.sv#L16

latsarbomba, 2021-02-11:
> Hi and super thanks for sponsoring, it's very nice since Github will sponsor me with an equal amount effectively doubling your sponsorship. Thank You 💖
>
> Yes, RaptorDMA is 100% compatible with the [...]

Hi ulf, i built out the firmware per your updates, but apperently its refusing to flash via ocd or the diligent method. I attempted to go to the raport site for flashing guides, but thats apperently lacking as well. Do you know how i can push the modified bitstream to the raptor? 

Thank you,

ufrisk, 2021-02-12:
My RaptorDMA does not have the additional JTAG port soldered on so I can't really try it.

I did a flash with OpenOCD a while back when I got it; I'll have to try it again and get back to you on this one.

latsarbomba, 2021-02-12:
> My RaptorDMA does not have the additional JTAG port soldered on so I can't really try it.
> 
> I did a flash with OpenOCD a while back when I got it; I'll have to try it again and get back to you on this one.

Thank you! I appreciate your time in this :)

latsarbomba, 2021-02-12:
Good evening ulf, I just wanted to let you know i was able to get it flashed via OCD, Thank you.

closed 2021-02-12 by latsarbomba

ufrisk, 2021-02-14:
Huge thanks for letting me know. I'm glad you managed to resolve the issue. Best wishes with your DMA attacking.

Berniokas, 2022-04-22:
Hello
Has anyone bought a RaptorDMA FPGA board from raptor-dma.com? Is it a reliable store or is it a scam? No any contact information in website, 9 boards in stock at this moment.

ufrisk, 2022-04-22:
@Berniokas RaptorDMA have actively decided not to support this project with a single cent since over a year since my project is not important for their sales. I've put in a lot of work into my tools and they are doing a lot of money out of it giving absolutely nothing back.

From what I hear Screamers will be restocked at LambdaConcept hopefully within a month if everything goes well.

Berniokas, 2022-04-25:
Thanks for the explanation, I'll waiting for LambdaConcept.

ufrisk, 2022-04-25:
Awesome, now lets hope there won't be any delays in he resupplying :) I haven't heard anything so I think it will be fine.

pineda89, 2023-05-28:
> Good evening ulf, I just wanted to let you know i was able to get it flashed via OCD, Thank you.

hello mate, i'm having exactly the same problem as you
Could you tell me how you got it? Thanks

blueeorx1, 2023-11-07:
@pineda89 @latsarbomba Hello guys i'm having also the same problem i'm ready to pay for helping
Thanks

btequila, 2024-01-13:
@latsarbomba Hello, ive got exactly the same problem as you had.

Can you please help me? Could pay a bit...

btequila, 2024-01-13:
@pineda89 @blueeorx1 does anyone have a solution now? Would pay a bit for helping... thanks :)

Angrypanda06, 2024-01-13:
I'll be happy to help yeah, just send me a dm and we'll get you sorted out