ufrisk/pcileech-fpga #87: Cannot Compile ScreamerM2 4.8 Source

mmollet, 2021-02-04:
Hi Ulf, i cant compile it, i always get error in the first step

Vivado% source vivado_generate_project.tcl -notrace
WARNING: [IP_Flow 19-395] Problem validating against XML schema: see 'xilinx:componentInstanceExtensions' : Bad end of element
CRITICAL WARNING: [IP_Flow 19-3389] Failed to import IP file 'C:/ScreamerM2/ip/bram_pcie_cfgspace.xci': Could not find IP definition in IP file 'C:/ScreamerM2/ip/bram_pcie_cfgspace.xci'. The IP file may be corrupt.

CRITICAL WARNING: [filemgmt 20-984] The source file 'C:/ScreamerM2/ip/bram_pcie_cfgspace.xci' cannot be added to the fileset 'sources_1'.
WARNING: [IP_Flow 19-395] Problem validating against XML schema: see 'xilinx:componentInstanceExtensions' : Bad end of element
CRITICAL WARNING: [IP_Flow 19-5097] Unable to determine VLNV from IP file; verify it has the correct syntax: C:/ScreamerM2/pcileech_screamer_m2/pcileech_screamer_m2.srcs/sources_1/ip/bram_pcie_cfgspace/bram_pcie_cfgspace.xci
ERROR: [Common 17-107] Cannot change read-only property 'synth_checkpoint_mode'.
Resolution: Please refer to Vivado Properties Reference Guide (UG912) for more information on setting properties

ufrisk, 2021-02-04:
which version of vivado are you running?

ufrisk, 2021-02-08:
I really do need this information to be able to look into this. Are you running Vivado 2020.2 when you're trying to build and still getting these errors?

mmollet, 2021-02-09:
sry, i used 2020.1, i didnt even know there was a 2020.2 version

ufrisk, 2021-02-09:
I now see that the readme said 2020.1. I have now updated this. Thanks for letting me know.

2020.2 is required so you'll need to update and the issue will go away. I'm closing this issue since it's now resolved.

closed 2021-02-09 by ufrisk