ufrisk/pcileech-fpga #92: Virtex®-7 FPGA VC707

sdanfa, 2021-08-10:
Hello,
How to make this work with my board, Virtex®-7 FPGA VC707?

Thanks.

ufrisk, 2021-08-10:
That is a very nice board :)

Easiest is probably to stick with the FT601 FMC daughter board I use in the AC701 and SP605 designs.

After that it should mostly be a matter of altering the PIN assignments in the .xdc file and re-generate the IP-blocks.

It's a bit of work to do this; but I hope it shouldn't be too hard.

Unfortunately I don't have the VC707 myself so I have no immediate plans on supporting it. If you get it up and running I could link to it though if you publish it on Github.

sdanfa, 2021-08-10:
> That is a very nice board :)
>
> Easiest is probably to stick with the FT601 FMC daughter board I use in the AC701 and SP605 designs.
>
> After that it should mostly be a matter of altering the PIN [...]

@ufrisk Yes, the board is very nice, thanks.

All I need to change the PIN's is reading the pdf documentation of the boards?
So I just check your boards PIN's like "C15 FMC1_HPC_LA10_N LVCMOS25 A18" and change it to "C15 FMC1_HPC_LA10_N LVCMOS18 M39" in the .xdc file?

I read documents from https://www.xilinx.com/support/documentation/boards_and_kits/ac701/ug952-ac701-a7-eval-bd.pdf
and https://www.xilinx.com/support/documentation/boards_and_kits/vc707/ug885_VC707_Eval_Bd.pdf ?

Thanks.

sdanfa, 2021-08-10:
Also do I have to edit PACKAGE_PIN only or IOSTANDARD  too in xdc file?

sdanfa, 2021-08-10:
I got stuck on "set_property LOC GTPE2_CHANNEL_X0Y7". VC707 have GTX instead of GTP. I can't find the right documentation about it in the PDF's.

sdanfa, 2021-08-10:
Can I set PCIE LANE 0 with the following code?
`    # PCIe Lane 0
    set_property LOC GTXE2_CHANNEL_X1Y11 [get_cells -hierarchical -regexp {.*pipe_lane\[0\].gt_wrapper_i/gtx_channel.gtxe2_channel_i}]`

ufrisk, 2021-08-11:
It may be that you'll have to regenerate the PCIe IP core.

Also check out the placement of the PCIe PINs in the .xdc file. Also the PIN I use for SYSCLK and make sure the frequencies are correct (you can convert them in clock conversion IP Core I have in the AC701 design.

I'll publish a small bug fix release in an hour or two that you may wish to base your design upon.

Good Luck :)

sdanfa, 2021-08-11:
> It may be that you'll have to regenerate the PCIe IP core.
>
> Also check out the placement of the PCIe PINs in the .xdc file. Also the PIN I use for SYSCLK and make sure the frequencies are correct [...]

Okay thanks for the help. :)