ufrisk/pcileech-fpga #96: Open Source DMA hardware????

huangfengye, 2021-12-23:
Hi, if anyone tried this??
https://github.com/FotatoPotato/Artix-Express-35

ufrisk, 2021-12-26:
Thanks. I was not aware. It seems like a device similar to the PCIeScreamerR04 but more geared towards development.

KengKengu, 2022-01-01:
Should the software work with a device like this?

https://rhsresearch.com/products/litefury

It's not on the list in the README but I don't see why it would not work.

ufrisk, 2022-01-01:
@KengKengu The software works with the LiteFury, but you'd need additional hardware for PCILeech to communicate with it over USB. I detail this in this blog entry: http://blog.frizk.net/2021/10/acorn.html

KengKengu, 2022-01-01:
Thank you very much, that was very informative and I am grateful. Unfortunately I am not sure if I can complete all the steps. The biggest issue I have is removing the 3.3V to 2.5V voltage regulator, as I don't have the required tools. 

Do you think there is any way I could remove it without the hot-air rework station?
Otherwise I will have to use the another way to communicate, which would be much slower.

ufrisk, 2022-01-01:
Maybe there are other USB modules; but my guess is that it would involve significant work to get it working. I'm not going to help with that. Unfortunately the Nitefury just isn't very suitable for PCILeech DMA accesses. You're probably better off purchasing a Screamer or EnigmaX1 board (when they get in stock).

olealgoritme, 2022-04-14:
The acorn blogpost was awesome! Will be trying this soon with my acorns:) 

One thing though, the link in the post doesn't exist:
https://github.com/ufrisk/pcileech-fpga/acorn/

ufrisk, 2022-04-15:
@olealgoritme Thank you for pointing this out. I've updated the link in the blog post to the correct project pages.

ufrisk, 2022-12-04:
I'm closing this thread due to old age. Nothing has changed with regards to open source hardware as far as I know.

But, there are some lower cost options available. One example is the DMANinja https://www.aliexpress.com/item/1005004823146314.html

closed 2022-12-04 by ufrisk