ufrisk/pcileech-fpga #97: Speed issue

latsarbomba, 2022-02-04:
Howdy Ulf, 
Currently trying to trouble shoot a speed issue on a couple different m.2 screamers running the 4.8 bitstream.
Tried the two versions of the command to check speed/fails ect and the results are always the same 

pcileech.exe -device fpga -v dump -out none
pcileech.exe -device fpga -v dump -out none -mmemauto
[image]

On a third test m.2 board  I get the expected results of 110/120 MB/s so im wondering if its an issue with the way the firmware was compiled on the problematic boards or if its an issue with the attacking pc. 

Is there any guidance you can share?

ufrisk, 2022-02-04:
it says it downgraded the algorithm to "tiny". it's slower. Most likely it's an issue with your target PC.

latsarbomba, 2022-02-04:
Stripped out all the extra ssd drives on the target pc and speeds stabilized, was losing my mind. 

Thank you Ulf.

closed 2022-02-04 by latsarbomba