ufrisk/pcileech-fpga #99: Recovering from a BAD flash

Hayden1996562, 2022-03-26:
I think I have somehow done a bad flash, is there a way to reset the m.2 screamer board, I have tried reflashing but it doesn't seem to bring it back to life. It still does recognize a usb connected to second pc but does not recognize host pc.

Hayden1996562, 2022-03-26:
USB driver already existed or was successfully installed.

ucrtbase.dll exists
        1 file(s) copied.
ucrtbased.dll Successfully copied.
ucrtbase_clr0400.dll exists
ucrtbase_enclave.dll exists
vcruntime140.dll exists
        1 file(s) copied.
vcruntime140d.dll Successfully copied.
vcruntime140_1.dll exists
        1 file(s) copied.
vcruntime140_1d.dll Successfully copied.
vcruntime140_clr0400.dll exists
ucrtbase.dll exists
ucrtbased.dll exists
ucrtbase_clr0400.dll exists
vcruntime140.dll exists
vcruntime140d.dll exists
vcruntime140_clr0400.dll exists

Starting pcileech test.

DEVICE: FPGA: ERROR: Unable to connect to USB/FT601 device [0,v0.0,0000]
PCILEECH: Failed to connect to the device.

Hayden1996562, 2022-03-26:
Also I have no lights on my m.2 screamer I think it may of died

ufrisk, 2022-03-27:
Are you able to connect to it over JTAG? It must be powered from the M2 side when you're doing it. Also take great care not to short circuit the board beneath the JTAG port, i.e. put an electric tape beneath or something.

It's rather impossible for me to say if it's fried or if it's just a bad flash. I really do hope you get it up and running. Please let me know how it goes.

ufrisk, 2022-04-26:
Unfortunately I don't think I can help more with this. I wish I could say otherwise but I guess things are as they are. I'm closing the issue.

closed 2022-04-26 by ufrisk