ufrisk/pcileech-fpga #100: LiteFury flashing problem

Berniokas, 2022-04-15:
Hello,
I had a problem with connecting to LiteFury with Digilent HS2 adapter for flashing Pcileech to it. Vivado Hardware manager can't detect HS2 and LiteFury board. I using original LiteFury JTAG cable. Digilent drivers are installed and in FT_Prog program HS2 cable is detected. Did I need to change any settings in FT_Prog? Current HS2 settings are attached in screenshots.  
Thank you in advance.
Best regards
[image]
[image]
[image]
[image]
[image]
[image]

Marek

ufrisk, 2022-04-15:
hm, for flashing with the HS2 you should not change anything in the FT_Prog. I never did ever. Make sure the HS2 is connected as I show here: https://lh3.googleusercontent.com/-2vfVlTj7pn4/YWbocqUV_7I/AAAAAAAAAa0/o-vWEZ2JPkwXZ60DdXDonRGgznqkvj1mgCLcBGAsYHQ/s1600/jtag-1.jpg and also that the LiteFury is powered by itself. It should be detected automatically by vivado when connecting to the HS2.

ufrisk, 2022-04-21:
I'm assuming this issue was resolved. Closing the open issue. If you still have any questions please let me know in this thread.

closed 2022-04-21 by ufrisk