ufrisk/pcileech-fpga #101: Find compatible hardware

Stoemal, 2022-04-21:
Hi,

Since the time PCILeech was developed, hardware became kind of unavailable and expensive.
The Screamer, Enigma and LiteFury are all out-of-stock and the [Spartan SP605](https://www.mouser.fr/ProductDetail/Xilinx/EK-S6-SP605-G?qs=rrS6PyfT74d4%252B96vRxt32Q%3D%3D&gclid=EAIaIQobChMI-pLWodCi9wIVxAIGAB2wHgd7EAAYASAAEgJuy_D_BwE) is now at about 1100€ (it was less than 500€ during your conferences).
I did a few research and found this [Artix-7 FPGA developement board](https://fr.rs-online.com/web/p/modules-de-developpement-fpga/1346478) that only costs 200€ and has the same FPGA used in The Screamer. The USB interface can only stream in USB 2.0, but speed in my case isn't that much of an issue.
I also found this [conference](https://www.blackhat.com/us-19/briefings/schedule/#picodma-dma-attacks-at-your-fingertips-15851) where he uses a PicoEVB with PCILeech even though PCILeech doesn't seem to have a support specific to the PicoEVB.
Again the PicoEVB has an Artix-7 FPGA (with 50K logic cells instead of 35K).
My guess is that most of the Artix-7 FPGA with a USB interface will work with PCILeech.
Am I right ?
Do you think the Artix-7 board I found could work with PCILeech ?
Do you have any idea on what available and affordable FPGA board I could use ?

Apart from that, amazing work with the PCILeech !! 
Have a good day !

ufrisk, 2022-04-21:
The hardware issue have been quite sad recently due to the global chip shortage and supply chain issues.

As far as I know LambdaConcept will have new stock of Screamers hopefully in the not too distant future.

For PCILeech I support USB with the FT601 chip or FT2232H chip. Other configurations are currently unsupported - i.e. you'll have to roll your own. It's what they did with the PicoEVB.

The board you're linking to is lacking PCI Express - so it will not work with PCILeech regardless of USB connectivity.

Stoemal, 2022-04-21:
Yes that's what I found out this morning while I was looking to interface M.2 to the FPGA.
Do you know if there would exist some adapter board in PCIe or M.2 that I could connect to the FPGA (I couldn't find one) ? 
Thanks a lot for the answers !

Stoemal, 2022-04-21:
No I meant for the [Artix-7](https://fr.rs-online.com/web/p/modules-de-developpement-fpga/1346478), but it's fine you helped me enough !
Thank you

ufrisk, 2022-04-21:
Nice to hear that. and once again I'm quite saddened by the current hardware situation, but there is not really much I can do about it. Best is probably to wait a month or so to see if the Screamers are restocked.

closed 2022-04-21 by ufrisk

Stoemal, 2022-04-22:
Yes I talked to them, they will have some stock back soon, but only for PCIe slots.

ufrisk, 2022-04-22:
Nice, and yes that seems to be the general trend unfortunately. But it's where most demand is it seems like.

There exists good adapters between M2 and PCIe though - but they'll always be a lot more clumsy than a pure M2 solution.

Stoemal, 2022-04-22:
I only found one [adapter](https://www.reichelt.com/fr/fr/delock-adaptateur-m-2-ngff-cl-m-m-le-emplacement-pci-express-delock-62584-p154304.html?PROVID=2810&gclid=EAIaIQobChMI-Pais-yk9wIVn4xoCR0RigzTEAQYBiABEgLXZPD_BwE) that I consider reliable. Do you have links or recommendations ?
And yes the semiconductor crisis is so annoying, soon we will have to fight for a few resistors...

ufrisk, 2022-04-22:
This one is more flexible from the Wifi-slot (key a+e). https://www.aliexpress.com/item/1005003809627656.html 

It have cable to grab 12V power from a SATA connector as well. Similar cables exists for other form factors or just as PCIe extender cables.

Stoemal, 2022-04-22:
Indeed, it looks nice too. 
Thanks for the link !

Stoemal, 2022-04-25:
Hi, It's me again
I was thinking, is the VT-DIO enabled for ports such as SODIMM and ExpressCard. Because if not, it would enable us to bypass this security ! Do you know if it is the case ? 
Also, I saw you using an Express Card during your conferences. Do you have a link to this PCIe to ExpressCard adapter ? Also if you know a PCIe to Thunderbolt adapter I would be very interested ! I found [this](https://fr.aliexpress.com/item/1005002473932082.html?spm=a2g0o.productlist.0.0.26045b33eWer60&algo_pvid=e88705a2-4d6c-4562-8974-24a911241ef6&algo_exp_id=e88705a2-4d6c-4562-8974-24a911241ef6-10&pdp_ext_f=%7B%22sku_id%22%3A%2212000020777892198%22%7D) but I would also need a PCIe to M.2 key A adapter. It makes a lot of adapters.

ufrisk, 2022-04-25:
ExpressCard is pretty much hot pluggable PCIe as-is. This works very well with PCILeech.

RAM slots are a completely different story. They are very high frequencies, also the memory controller scrambles memory for performance reasons. It would be super hard (and super expensive) to intercept and modify contents on the fly.

The ExpressCard adapter is pretty much the same as the one I use. Pair it with a mini-PCIe to PCIe adapter and you're good to go.

ufrisk, 2022-04-26:
Yes, I know about Thunderspy. In Thunderspy they reflash the Thunderbolt controller to disable security. Then they use PCILeech to perform the actual DMA attack.  PCILeech had already supported Thunderbolt a long time before Thunderspy came around - but only if Thunderbolt security had been manually disabled by the user prior to the attack.

Stoemal, 2022-04-27:
Yes but you can't disable Thunderbolt security if the user has put a BIOS password. At this moment Thunderspy comes handy right ? 
I'm doing pentesting, I'm trying to attack the DMA side.

ufrisk, 2022-04-28:
Absolutely, Thunderspy is a super nice way to enable DMA access over Thunderbolt without having prior access. When doing pentests it makes total sense.

Stoemal, 2022-04-29:
Nice thank you !

Stoemal, 2022-05-05:
Hi, 
It's me again ! 
Even though I don't have a Screamer yet, I am trying to build the PCILeech software. I am having some trouble to build it.
From what I understand, I must first compile LeechCore to be then able to build MemProcFS and PCILeech.
When I build LeechCore, I am having build errors about a python.h file VS2022 can't open.
When I build MemProcFs, I don't VS can't build since LeechCore is not present in the project. But I don't know which file or folder to place, and I don't know where to place it (I thought it might be a .dll or .exe or .vcxproj)
When I try to install the dokany lib, I must install a WDK module, but the WDK for VS2022 isn't developed yet. And when I try to build it with VS2019, it has error of a MSVC v143 not present (it is a lib only in VS2022 from what I understand).
When I try to build PCILeech, LeechCore and MemProcFs are not present (logical so far), but I don't know how to add them to the solution.
It's is not even including the LeechFPGA and PushPin installations.
Also you can install MemProcFs with pip ? Can you do the same for the other packages ? 
I read almost all the repo but can't find precisions, could you help me with the installation please ?  
If you have videos explaining or threads detailing the installation it would still help a lot !

I installed Windows 10 in a VM (just for testing the build not to perform a real attack)
I installed VS2019 and VS2022 with the C++ and Python modules + the WDK for VS2019
I also installed the latest version of python and added it to the PATH

Thanks again

ufrisk, 2022-05-05:
1. In order to build the Python module for LeechCore you do need to have Python installed. It's an optional Visual Studio feature. Install Python via the Visual Studio installer first.
2. In order to build MemProcFS you'd need to have LeechCore built first. Place LeechCore and MemProcFS in the same parent directory, example `C:\Build\LeechCore` and `C:\Build\MemProcFS`
3. You do not need to build Dokany from scratch. MemProcFS does not require it. Use their binaries. If wishing to build Dokany from scratch please consult their project - I'm not going to support that. But regardless you'd need to somehow get your Dokany kernel driver signed properly if wishing to build Dokany from scratch.
4. PushPin is a separate 3rd party project. Please consult the PushPin project for that.
5. `pip install memprocfs` should work fine. It should install the `leechcorepyc` packet as a dependency.
6. As for using the Python library please consult the guide at: https://github.com/ufrisk/MemProcFS/wiki/API_Python there is a video there as well.

Good Luck and best wishes with everything 👍

ufrisk, 2022-05-09:
Awesome and thanks for the update 👍  I'm happy to see it seems like it's working. From what I heard hardware should be back in stock real soon.

Stoemal, 2022-05-10:
Yes I exchanged with Lambda, they'll have hardware this week !!

ufrisk, 2022-05-11:
Yes, they released it today at shop.lambdaconcept.com

Best wishes with your DMA attacking, and please let me know how it goes.

Stoemal, 2022-05-18:
> Yes, I know about Thunderspy. In Thunderspy they reflash the Thunderbolt controller to disable security. Then they use PCILeech to perform the actual DMA attack. PCILeech had already supported [...]

I had a question about your above answer, when you say that PCILeech supports Thunderbolt. Do you mean we can perform DMA attacks by simply connecting to the Thunderbolt/USB-C connector ? If so, do you know a connector that could adapt Thunderbolt to PCIe or M.2 ? I thought about [this](https://www.amazon.fr/Bo%C3%AEtier-Externe-FIDECO-Adaptateur-Soutenir/dp/B07TSBR114/ref=asc_df_B07TSBR114/?tag=googshopfr-21&linkCode=df0&hvadid=353825338977&hvpos=&hvnetw=g&hvrand=17929808171794851236&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9056492&hvtargid=pla-801692183482&psc=1&tag=&ref=&adgrpid=69466803497&hvpone=&hvptwo=&hvadid=353825338977&hvpos=&hvnetw=g&hvrand=17929808171794851236&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9056492&hvtargid=pla-801692183482#customerReviews), but it seems to just be USB3 in a USB-C connector.

ufrisk, 2022-05-18:
I use this: https://www.aliexpress.com/item/1005002680230415.html

But there exists many NVMe enclosures as well, but I'm rather hesitant of recommending any of those since it's as you already found it that many of those only use USB3 and not Thunderbolt.

Stoemal, 2022-05-18:
Ooh I see, I found it at first but it was expensive and since I wasn't sure of the product I tried to find something cheaper (which I couldn't find as you can see). 
Thanks !

Stoemal, 2022-06-29:
Hi Ulf, 
It's me again. I now have all I need to perform DMA Attacks. 
But I am having some trouble with PCILeech. 
I'm having the error : "Unable to retreive required Device PCIe ID [4,v4.10,0000]
I think I added well the FTDI drivers because befaore adding them I had the error "Can't load FTDI Drivers" or so.
I tried installing the one from FTDI as well, but it didn't change anything. 
The Screamer seems well powered, but is not connected to a victim laptop yet (only the analysis laptop).
I believe it should still display me some message like "PCIeScreamer identified" like in [this page](http://docs.lambdaconcept.com/screamer/getting_started.html#run-pcileech)
The analysis laptop I am using is running on Windows 10 but is fairly old (Latitude E6430)
Some of my PCI drivers are not up to date, do you think it could be the cause of the problem ? I tried to update them but didn't succeeded. I'll try again tomorrow.
Do you have a possible solution to this problem ? 
Is there a sort of cheat sheet of what to try depending on the error message ? 

Thanks, 
Have a good day

ufrisk, 2022-06-29:
You're able to communicate with the device. It retrieves that it's of firmware version 4.10.

It's however not assigned a PCIe ID bus:dev:num hence the 0000.

Basically there is no PCIe link. This can be for multiple reasons, but I'm not really able to detect why it's not assigned an ID. So there is very little I can do about the error message. It says Unable to retrieve required PCIe ID which is correct. And I'm not really able to know why that is unfortunately.

Try starting the computer with the board already inserted, disable VT-d and such in the BIOS, if using Thunderbolt disable security etc.

ufrisk, 2022-07-03:
Huge thanks for the update and good luck with everything! 👍

Stoemal, 2022-07-18:
Hi, 
I ordered the PCIe to Thunderbolt adapter from Aliexpress, I am now waiting for it to arrive.
In the mean time I found other American and European manufacturers/retailers. 
Here are the links if you're interested. 
https://www.alternate.fr/Sonnet/Allegro-carte-et-adaptateur-dinterfaces-Interne-USB-3-2-Gen-1-(3-1-Gen-1)-Contr%C3%B4leur-USB/html/product/1690832
https://www.sonnetstore.com/collections/thunderbolt-upgrade-cards
https://www.platinium.fr/store/fr/composants-materiel/593-carte-d-upgrade-thunderbolt-3-pour-echo-express-iii-d-et-iii-r-732311012273.html

Have a good day :)

Stoemal, 2022-07-22:
It's me again, 
I am now trying to perform a DMA attack through Thunderbolt, but I am getting some errors. 
The error I am having is : Failed to connect to the device 
I changed the Security Levels to SL0 and disabled every possible security in the BIOS.
I think this error comes from a hardware problem concerning my Thunderbolt adapter (it arrived !).
I am using this hardware setup : 

The Green PCB is the Thunderbolt adapter, The Red PCB is the Screamer, The Black PCB is a card to interface the two previous boards.

[image]

I tested the Black PCB and it seems 4 capacitors are fried, I suppose the chips next to them are fried too. And the chips are impossible to find. 

[image]

I bought the Black PCB to supply the Screamer in 12V and to have a sturdy mechanical assembly. But since it doesn't work, I thought about directly plugging the Screamer in the adapter such as below : 

[image]

I'll just have to solder the 12V power supply directly to the Screamer. There is also the R119 that I believe I shouldn't shunt since it would connect the 5V of the adapter to the 12V of the Screamer (I suppose It will fry the adapter). 
I am not sure about this configuration (whether it could fry the Screamer or the adapter). 
The only PCIe pins connected to the chip on the Black PCB (apart from the 12V and GND) are PCIe Reset Signal and Hot Plug Presence Detect. The other pins are in direct connection between the PCIe Ports.

What is your configuration like ? Should I order a new Interfacing board ? Do you think the problem could come from somewhere else ? 

If you want me to stop writing in this issue, I can open a new one or even change of communication medium. 
If you already answered this, I am sorry I'll just search better in the issues section. 
And if I am too insistent, don't hesitate to tell me !

Anyway 
Thanks a lot for your help 
Have a good day !

ufrisk, 2022-07-23:
For thunderbolt I most often use the ScreamerM2 (which is no longer on sale) since it only require 3.3V.

When I use the Screamer Squirrel I use this:
https://www.computersalg.se/i/3700589/navilock-str%c3%b6madapter-ac-100-240-v
pared with a bitcon mining adapter something like this:
https://www.aliexpress.com/item/1005003483136650.html

Stoemal, 2022-07-24:
I only have the Squirrel and I don't think I could obtain the Screamer in M.2. Do you have advices to use the Squirrel on Thunderbolt ? How do you use the Screamer M.2 on Thunderbolt ?

ufrisk, 2022-07-24:
ScreamerM2 is sold out and discontinued, but it will work directly in your thunderbolt adapter.

As far as the PCIe Squirrel goes, I just posted the two hardware that will solve your issue above. Just connect the bitcoin mining pcie extender to the Thunderbolt adapter you already got. And then use the power unit to power it with 12V.

Stoemal, 2022-07-24:
But how do I connect the Screamer to the Bitcoin mining adapter ? The PCIe slot will already be taken by the Thunderbolt adapter. And the Screamer needs to have a PCIe connection to the adapter to work, but there will only be the USB3 connector left.

ufrisk, 2022-07-24:
You insert the PCIe male connector of the bitcoin adapter into the thunderbolt adapter. You insert the Screamer in the bitcoin adapter female side.

Stoemal, 2022-07-26:
Oh I see ! 
I always thought these USB2 connectors were for USB2 communication, but looking at it closely, it is some PCIe x1 you're right. 
Thank you, you're helping me so much ! 

Excuse me for the late response, I read your answer in the hour it arrived, but I forgot to answer you back.

Stoemal, 2022-08-05:
I have made great advances ! I attacked with PCILeech in Thunderbolt. At first it didn't work. But then I retrieved the physmemmap.txt (Like told in [this issue](https://github.com/ufrisk/pcileech/issues/158) and the [wiki](https://github.com/ufrisk/LeechCore/wiki/Device_FPGA_AMD_Thunderbolt)) file thanks to MemProcFS and it worked ! I even arrived to disable the Thunderbolt security with Thunderspy method. 
I was wondering if it was possible to retrieve pysmemmap.txt directly via Thunderbolt. Because yet I am forced to pass through M.2 with VTd-IO and VBS disabled (in this case it is not relevant to attack in Thunderbolt since I already have access to the BIOS and DMA via M.2). 
Here is my error code : 
[image]

Thank you

ufrisk, 2022-08-05:
I'm not aware of a way that will work 100%. Two partial methods come to mind though.

1) Purchase the same model of computer. The memory map will be the same or very similar (if having the same hardware config and same amount of RAM).

2) Try your luck with `memprocfs.exe -memmap auto -max <max_guess_address>` - it will start memprocfs with conservative settings and if it works you may be able to retrieve the memory map from M:\sys\memory\physmemmap.txt

Stoemal, 2022-08-07:
Thanks a lot, I'll try this out !

Stoemal, 2022-08-08:
I tried today to use the above command, and it works very well. Basically an address that has the same order of magnitude as the physical memory present on the computer, without being larger works ! It is not that hard to guess. 
I now have another question, I am trying to dump the SAM credentials using the MemProcFS plugin.
At first I tried to do so by mounting on the live RAM. Like this

[image]

(I also tried without the -pythonpath since I added python to the PATH, but I tried it anyway)
And the debug tells me the plugins were well loaded

[image]

But when i look at the py folder it is totally empty 

[image]

Then I saw your twitter post that says you can dump the secrets from a memory dump file.

[image]

So I dumped the memory and opened it with MemProcFS, but again the py folder is empty. 
I believe I installed wrongly the plugins or python. It may be a stupid error, but right now I am stuck. 
I you know what my error is, it would be helpful. Otherwise don't bother.
I am trying to test DMA as extensive as possible (Thunderbolt, MemProcFS plugins etc...)

Thanks
Have a good day

ufrisk, 2022-08-08:
have you unzipped all the files from the latest binary release to the MemProcFS folder?

also if you have previously installed the memprocfs via python pip it's likely that you'd need to upgrade the memprocfs package as well since it may have bad interactions with this.

Stoemal, 2022-08-09:
I reinstalled python, memprocfs, leechcore, pypykatz module 1 and 2, added them to the path, downloaded the latest MemProcFS binaries, extracted and placed the plugins in every possible MemProcFS folders. And now it works !!
I believe I didn't used the latest binaries or so. 
Anyway, now it works nicely.
Thank you a lot !