ufrisk/pcileech-fpga #102: Can PCILeech FPGA use PCIe x4?

mrtamyan, 2022-05-04:
Hello @ufrisk,

Can PCIeScreamerR04 work with x4 pcie lane width?  if yes, what effort is required?  I am trying to do it myself, I will really appreciate if you can give me some details what parts are required potential changes.

mrtamyan, 2022-05-05:
I saw it in git log that was about version 3.
```
    output  [3:0]   pci_exp_txp,
    output  [3:0]   pci_exp_txn,
    input   [3:0]   pci_exp_rxp,
    input   [3:0]   pci_exp_rxn,
```
Also I can not find any data about PCIeScreamerR04 pinout, but I found some as comments in pcileech_screamer_m2.xdc. Are they all correct? 
```
#set_property PACKAGE_PIN A3 [get_ports {pcie_rx_n[1]}]
#set_property PACKAGE_PIN A4 [get_ports {pcie_rx_p[1]}]
#set_property PACKAGE_PIN F1 [get_ports {pcie_tx_n[1]}]
#set_property PACKAGE_PIN F2 [get_ports {pcie_tx_p[1]}]

#set_property PACKAGE_PIN C3 [get_ports {pcie_rx_n[2]}]
#set_property PACKAGE_PIN C4 [get_ports {pcie_rx_p[2]}]
#set_property PACKAGE_PIN D1 [get_ports {pcie_tx_n[2]}]
#set_property PACKAGE_PIN D2 [get_ports {pcie_tx_p[2]}]

#set_property PACKAGE_PIN G3 [get_ports {pcie_rx_n[3]}]
#set_property PACKAGE_PIN G4 [get_ports {pcie_rx_p[3]}]
#set_property PACKAGE_PIN B1 [get_ports {pcie_tx_n[3]}]
#set_property PACKAGE_PIN B2 [get_ports {pcie_tx_p[3]}]
```

mrtamyan, 2022-05-05:
Or could you release your x4 design in any branch if it is possible, please?

ufrisk, 2022-05-05:
Hi, yes, changing the PCIe lanes is one piece of the puzzle though. To continue to use the current design you'd need to clock the core at 250MHz though which will make timing requirements to fail miserably. To use x4 you'd need to use the 128-bit transaction interface which is a large re-design of the whole thing which affects pretty much everything.

There is some work for me preparing the x4 design for release and since there is currently no performance benefit or project benefit in it I'm not really keen on spending time on it. Please understand that this is free open source and I don't get paid for this. My time is currently better put elsewhere.

Things may change in the future though, and it may be that I will base some new designs on the x4 design, but it's not something I currently have any plans for.

ufrisk, 2024-01-17:
I've updated the AC701 firmware to x4 in v4.13. It should work well. But there is no speed increase due to the USB bottleneck.

I haven't updated the ScreamerM2/ScreamerR04 firmware since its not giving a performance boost over x1, and also there is some additional (unpaid) work for me doing that involved. But it should be fairly easy to adapt the AC701 x4 design to ScreamerM2/ScreamerR04 as well.

I'm closing this issue due to old age.

closed 2024-01-17 by ufrisk