ufrisk/pcileech-fpga #104: Extract Firmware

Remashes, 2022-06-12:
Hello, I would like to know if it is possible to extract firmware from lambdaconcept screamer using openOCD.

ufrisk, 2022-06-12:
I think it should be possible, it's not encypted/protected in any way. But I've never done it or tried it with OpenOCD. I won't be able to help with this, but I'm sure there are some info if you google for it. Please let me know if you are successful :)

povauboin, 2022-06-21:
Yes but what would be the goal ? The firmware flashed on the screamer is here:
https://github.com/ufrisk/pcileech-fpga/tree/master/PCIeSquirrel#releases--version-history
https://github.com/ufrisk/pcileech-fpga/tree/master/ScreamerM2#releases--version-history

BG7YWL, 2022-08-20:
The firmware can be extracted from the FLASH chip.

ufrisk, 2022-08-21:
I'm closing this issue since I believe the question has been answered.

closed 2022-08-21 by ufrisk