ufrisk/pcileech-fpga #105: CFGSPACE_STATUS_REGISTER_AUTO_CLEAR

Brad-ADV, 2022-07-02:
Thanks for your work on this project.

Question on the CFGSPACE_STATUS_REGISTER_AUTO_CLEAR flag - does this need to be changed to a 1 to enable this feature, or is it set (disabling aborts) by default? 

Thanks

ufrisk, 2022-07-03:
Hi,

It would need to be enabled. It's disabled by default. If you wish to have it on at all times please change 0 to 1 on this line: https://github.com/ufrisk/pcileech-fpga/blob/cf0d80c60abb74def69fc9550f0515e59a220805/PCIeSquirrel/src/pcileech_pcie_cfg_a7.sv#L207

You'd need to rebuild the firmware and reflash after this.

It's also possible to change this value via an API call, but it will clear upon reboot or power-down. Please let me know if you wish to do it this way instead and I'll look into some sample code for this. I'm currently away from home for a few days so it will be hard to test it though (i.e. no access to hardware).

Please let me know if this answers your question and you're able to resolve the issue.

Brad-ADV, 2022-07-03:
Thank you for your quick reply.
I can modify that line and rebuild.

One more question if you don't mind:

Can you please just confirm that the correct steps to change the config space is to change line 268 on pcileech_fifo.sv to:

rw[203]     <= 1'b0;   

and then simply fill out the pci space in pcileech_cfgspace.coe and build the bitstream?

Thanks.

ufrisk, 2022-07-04:
Yes, exactly. Then build the bitstream.

If you have already built it once and are changing the `.coe` file in the project (and are doing a partial build) the change of the config space won't take effect though. You'd need to re-generate the `bram_pcie_cfgspace` core. But if doing it from scratch there shouldn't be any issues. Also change the DSN according to my build instructions.

Verify that the result is as you wish with SIV64 (or `lspci -xxxx` on Linux).

ufrisk, 2023-03-12:
Closing issue due to old age.

closed 2023-03-12 by ufrisk