ufrisk/pcileech-fpga #107: FPGA SP605 + ftdi usb3 ft601-b device isn't detected on my HOST system

sridhaleech, 2022-08-03:
Hi Frisk ,

I have tried FPGA enablement & unable to make it work USB3 ftdi ft601-b device with FPGA SP605 .
== > (https://github.com/ufrisk/pcileech-fpga/tree/master/sp605_ft601)

**for build** 
1.Install Xilinx ISE Development Environment.
2.Open ISE Design Suite 64-Bit Command Prompt.
3.Run build.bat to generate Xilinx proprietary IP cores and build bitstream.
Finished !!!

I'm getting some FIFO generation error because PCIleech source file (Downloaded from Github ) was generated FIFO with 9.3v + ISE14.7 version.

I'm using currently FIFO generator 8.4 ver with ISE 13.4 version.

My need is to do DMA read/write with FPGA SP605 + ftdi usb3 ft601-b on my system.

How to make it work FPGA SP605 + ftdi usb3 ft601-b device 

Past one month i'm struggling to enable FPGA with  ftdi usb3 ft601-b device  and ISE 14.3 +FIFO generator 8.4

Pls kindly advise me ,how to do it .

Thanks and advance .

sridhaleech, 2022-08-04:
Sorry man and thanks for support .

I will check with Xilinx team ,Can use 14.7 version instead of 13.4 .

Yes.then i will try to do it .

ufrisk, 2022-08-04:
Thanks. Or just use the pre-built version I have downloaded unless you'd need to make any changes.

closed 2022-08-04 by ufrisk

sridhaleech, 2022-08-04:
Hi,
I just tried with 14.7 ISE and while doing build noticed some error,during  " Generated programming file.
[image]
Any thing to do modification before build or ?
I just followed "Build" steps. Downloaded FPGA SP605_FT601 Files from below link and Just run build with 14.7 ISE 64-bit CMD window.
 
"https://github.com/ufrisk/pcileech-fpga/tree/master/sp605_ft601"

ufrisk, 2022-08-04:
great that the build worked. I assume you flashed it successfully too? There are no errors, just some warnings. It should work fine. Or just use the pre-built version.

sridhaleech, 2022-08-05:
Hi Frisk , 

After ran "Flash.bat” file & Flashed successfully. [image]

While flashing FPGA card jumper setting is  --> [image]
 
After flashed, The FT 601 USB was detected under device manager in my host system. at the time,  I just tried USB "PCIleech :Failed to detect " error came. 

So Then i changed Jumper setting as default based Xilinx and check FTDI601 device enumeration & its disappeared in device manger on my HOST system.

Now what I need to do here any isolation data can provide ?
 
And what is the  pre-built version ? how to do it, is there any steps to follow ?

sridhaleech, 2022-08-05:
Can re-Open the this issue .

sridhaleech, 2022-08-08:
Hi Frisk , Could you pls help me ?

ufrisk, 2022-08-08:
There is no need for a driver on the host system. DMA should work anyway.

sridhaleech, 2022-08-08:
Hi Frisk,

That Yellow bang was in target system. I have connected SP605 on my target system.
and now getting ethernet yellow bang on my target system. whether do we need install driver any for Target system or no need ?
I know, For HOST system will install only FT601 driver .

sridhaleech, 2022-09-07:
Hi frisk .
Can re-open this issue ?
Kindly update previous ask ?