ufrisk/pcileech-fpga #108: Need clarification for Access mode for SP605/FT601 device connected

sridhaleech, 2022-08-11:
Hi frisk,

I have noticed that "Access mode" DMA (hardware only) and Normal . what is the difference between this ?

I recently downloaded PCIleech binaries files and connected  64bit SP605 /FT601 device. While dumping the memory,We seen always --> Access mode is Normal.

While running with my old pcileech binaries files . Getting below as pic. ( "Access mode" DMA (hardware only))

[image]

While running with newly downloaded pcileech binaries files . Getting below as pic. "Access mode" is Normal

[image]

Why is this different results shown here  ?

ufrisk, 2022-08-19:
I changed the description text. It's no difference.

closed 2022-08-19 by ufrisk

sridhaleech, 2022-08-22:
Ohh! Thanks Frisk.
So we run the test with "Access mode : Normal" right  ?