ufrisk/pcileech-fpga #109: Need help for Screamer PCIe Squirrel card

sridhaleech, 2022-08-25:
Hi Frisk,

Actually, i need some clarification about this "Build ".
We have purchased ,Screamer PCIe Squirrel card last week along with USB-C cable and female USB-C to Type -A connecter.
"https://shop.lambdaconcept.com/home/50-screamer-pcie-squirrel.html"

Now my question is --> Do you need to purchase the Xilinx Vivado WebPACK 2021.2 or later. ?

Or is there any free version & we can download from Xilinx Website ?

Because, i need to flow the below steps for build to run on Xilinx Vivado WebPACK software in Win10 HOST system.

Building:
1. Install Xilinx Vivado WebPACK 2021.2 or later.
2. Open Vivado Tcl Shell command prompt.
3. cd into the directory of PCIeSquirrel (forward slash instead of backslash in path).
4. Run source vivado_generate_project.tcl -notrace to generate required project files.
5. Run source vivado_build.tcl -notrace to generate Xilinx proprietary IP cores and build bitstream.
6. Finished !!!

jacobhamblin, 2022-08-25:
Don't need to buy anything. I followed those steps using the software [here](https://www.xilinx.com/support/download.html)

first download link on this page - you can ctrl + f: Xilinx Unified Installer 2022.1: Windows Self Extracting Web Installer

sridhaleech, 2022-08-25:
Thanks a lot.. jacobhamblin. will try out it.

sridhaleech, 2022-08-25:
While installing "Xilinx Unified Installer 2022.1: Windows Self Extracting Web Installer" we have noticed that, there are many option to select. Can you pls help to confirm which option need to select ?

Picture here : 
[image]

ufrisk, 2022-08-25:
1. no need to flash this device. it comes pre-flashed with pcileech bitstream from the start.
2. if you wish to build anyway please follow instructions and install Xilinx **Vivado**.

best wishes with this.

sridhaleech, 2022-08-26:
Sure & big thanks for update.Frisk...

closed 2022-08-26 by sridhaleech