ufrisk/pcileech-fpga #110: PCILEECH: Failed to connected

sridhaleech, 2022-09-07:
Hi Frisk,

Need help.

Actually ,i brought this new Screamer PCIe Squirrel FPGA card and i connected to my windows target system.

i'm unable to read the memory due to **PCILEECH Failed to connect to the device** . attache below image.

[image]

In windows ,Squirrel FPGA card getting enumerated with ethernet controller yellow bang and now how to install the driver for that ?

From my target system yellow bang image below.

[image]

Because that PCIleech software is not executed. Kindly advise how to install driver and make it work pcileech software in WinOS level.(target )
=======
For example, Will install driver for USB3380 card in Windows OS using USB3380 Flash file.
 [usb3380_flash](https://github.com/ufrisk/pcileech/tree/master/usb3380_flash)

ufrisk, 2022-09-10:
Please download [FTD3XX.dll](https://www.ftdichip.com/Drivers/D3XX/FTD3XXLibrary_v1.3.0.4.zip) from FTDIchip and place it alongside pcileech.exe

also `pcileech.exe display -min 0x1000 -device fpga -v -vv` may give you extended error information.

The Squirrel does not required the USB3380 drivers. It requires the [FTD3XX.dll](https://www.ftdichip.com/Drivers/D3XX/FTD3XXLibrary_v1.3.0.4.zip) and also a kernel driver which will be installed automatically from Windows update if your PC is connected to the Internet at first use/connection of the Squirrel.

ufrisk, 2022-12-04:
I'm closing this issue due to old age. Also, I hope I've already answered your question in the answer above and that you already have it working.

closed 2022-12-04 by ufrisk