ufrisk/pcileech-fpga #111: Need clarification for mismatch  "pages read" during DMA

sridhaleech, 2022-09-15:
Hi Frisk,

We have noticed that when wo do DMA with as first command  using " pcileech.exe dump -min 0x0 -max 0xfffff -force -v" via PCIleech Spartan  SP605 or Squirrel card. Getting 160 pages only . Pls refer the below screenshot.

[image]

This is 1 MB file range (  0x0 -max 0xfffff ) .This 1MB size should be 256 pages. But we are getting only 160 pages.

Is this issue or any other ? 

Same command running successfully with USB3380 card ( its shown 256 pages)

Can help me to fix ?

**And also Can we use the latest pci Leech Binaries files ( pcileech_files)  for all leech card ?  like ( USB 3380,Spartan SP605 , Squirrel  Xilinx Card) **

Thanks / Sridhar

ufrisk, 2022-09-15:
It says 100% of the dump is failed. There is a special unreadable memory region between 0xA0000 and 0x100000 that will be avoided. I guess that's why you see 160 failed pages only in the first dump.

sridhaleech, 2022-09-17:
Okay and thanks for the updated

closed 2022-09-17 by sridhaleech