ufrisk/pcileech-fpga #115: Hi can you send out a version that supports kc705, I really need your help.

uk0, 2022-12-28:


uk0, 2022-12-28:
I modified the constaints and re-modified the voltage 2.5-> 1.8, passed the compilation, and generated the bin flash successfully, but it cannot be started. Now I guess what should be the cause of the reset.

ufrisk, 2022-12-29:
It's just very hard for me to tell what's wrong remotely without having my own KC705 to test on :(

You'll have to try to do some debugging to see what may be causing the issues.

FreshPizza, 2023-03-12:
I was curious so I ended up implementing this for a KC705 I had available. There's several revisions of the board and I have the 1.1/1.2 revision, and some of these boards may have different QSPI ICs (so flashing may fail, but it should print the detected part on the board).

I also couldn't quite get 4.11 working on the KC705 or AC701 either, not sure why, but didn't feel like troubleshooting that so I just used version 4.10

Dump (top is HPC, bottom is LPC - identical perf as expected)
[image]

DIP switch for SPI:

[image]

Memory Configuration Device (again, you may have a different device on your board):
`mt25ql128-spi-x1_x2_x4`

XDC for pcileech-FPGA **4.10**
```
#################################################
####  HPC definition, default FMC connector  ####
#################################################

# FMC_HPC_LA10_N
set_property PACKAGE_PIN C30 [get_ports ft601_rst_n]
# FMC_HPC_LA00_CC_P
set_property PACKAGE_PIN C25 [get_ports ft601_oe_n]
# FMC_HPC_LA00_CC_N
set_property PACKAGE_PIN B25 [get_ports ft601_rd_n]
# FMC_HPC_LA08_P
set_property PACKAGE_PIN E29 [get_ports ft601_wr_n]
# FMC_HPC_LA07_P
set_property PACKAGE_PIN E28 [get_ports ft601_rxf_n]
# FMC_HPC_LA07_N
set_property PACKAGE_PIN D28 [get_ports ft601_txe_n]
# FMC_HPC_LA08_N
set_property PACKAGE_PIN E30 [get_ports ft601_siwu_n]
# FMC_HPC_LA15_N
set_property PACKAGE_PIN B24 [get_ports {ft601_be[0]}]
# FMC_HPC_LA15_P
set_property PACKAGE_PIN C24 [get_ports {ft601_be[1]}]
# FMC_HPC_LA09_N
set_property PACKAGE_PIN A30 [get_ports {ft601_be[2]}]
# FMC_HPC_LA09_P
set_property PACKAGE_PIN B30 [get_ports {ft601_be[3]}]
# FMC_HPC_LA32_N
set_property PACKAGE_PIN C21 [get_ports {ft601_data[0]}]
# FMC_HPC_LA33_N
set_property PACKAGE_PIN H22 [get_ports {ft601_data[1]}]
# FMC_HPC_LA32_P
set_property PACKAGE_PIN D21 [get_ports {ft601_data[2]}]
# FMC_HPC_LA33_P
set_property PACKAGE_PIN H21 [get_ports {ft601_data[3]}]
# FMC_HPC_LA30_N
set_property PACKAGE_PIN C22 [get_ports {ft601_data[4]}]
# FMC_HPC_LA31_N
set_property PACKAGE_PIN F22 [get_ports {ft601_data[5]}]
# FMC_HPC_LA30_P
set_property PACKAGE_PIN D22 [get_ports {ft601_data[6]}]
# FMC_HPC_LA31_P
set_property PACKAGE_PIN G22 [get_ports {ft601_data[7]}]
# FMC_HPC_LA28_N
set_property PACKAGE_PIN C16 [get_ports {ft601_data[8]}]
# FMC_HPC_LA29_N
set_property PACKAGE_PIN B17 [get_ports {ft601_data[9]}]
# FMC_HPC_LA28_P
set_property PACKAGE_PIN D16 [get_ports {ft601_data[10]}]
# FMC_HPC_LA29_P
set_property PACKAGE_PIN C17 [get_ports {ft601_data[11]}]
# FMC_HPC_LA24_N
set_property PACKAGE_PIN A17 [get_ports {ft601_data[12]}]
# FMC_HPC_LA25_N
set_property PACKAGE_PIN F17 [get_ports {ft601_data[13]}]
[... 225 lines cut ...]
#set_property PACKAGE_PIN V2 [get_ports {pcie_tx_p[6]}]
#set_property PACKAGE_PIN Y6 [get_ports {pcie_rx_p[6]}]
#set_property PACKAGE_PIN V1 [get_ports {pcie_tx_n[6]}]
#set_property PACKAGE_PIN Y2 [get_ports {pcie_rx_n[6]}]

#set_property LOC GTXE2_CHANNEL_X0Y0 [get_cells {i_pcileech_pcie_a7/i_pcie_7x_0/inst/inst/gt_top_i/pipe_wrapper_i/pipe_lane[7].gt_wrapper_i/gtx_channel.gtxe2_channel_i}]
#set_property PACKAGE_PIN Y2 [get_ports {pcie_tx_p[7]}]
#set_property PACKAGE_PIN AA4 [get_ports {pcie_rx_p[7]}]
#set_property PACKAGE_PIN Y1 [get_ports {pcie_tx_n[7]}]
#set_property PACKAGE_PIN AA3 [get_ports {pcie_rx_n[7]}]

#PCIe clock
set_property PACKAGE_PIN U7 [get_ports pcie_clk_n]
set_property PACKAGE_PIN U8 [get_ports pcie_clk_p]

create_clock -name pcie_refclk_p -period 10.0 [get_nets pcie_clk_p]

set_property BITSTREAM.CONFIG.SPI_BUSWIDTH 4 [current_design]
set_property BITSTREAM.CONFIG.EXTMASTERCCLK_EN div-1 [current_design]
set_property BITSTREAM.GENERAL.COMPRESS TRUE [current_design]
set_property BITSTREAM.CONFIG.SPI_FALL_EDGE YES [current_design]

set_property CONFIG_VOLTAGE 1.8 [current_design]
set_property CFGBVS GND [current_design]
```

ufrisk, 2023-03-12:
Thanks for this info 👍 

It's a bit worrisome that the AC701 v4.11 isn't working though. I burnt mine some months ago (and since it's kinda pricey I don't feel like paying for a new one) so I kinda compiled in the new changes blindly. Must have missed something...

FreshPizza, 2023-03-15:
looks like it's the STARTUPE2 primitive that's doing it, after reverting this change from the 4.11 commit it seems to work fine on the AC701/KC705 https://github.com/ufrisk/pcileech-fpga/commit/996095fb612485ba25b4b49b9bc5078d84e24f95#diff-07e6827f4a38c28117b22c276c6e44fd34c25e1d99c087156283edd62e3e9238L14-R14

ufrisk, 2023-03-17:
I made the update to the AC701 sources and pre-built bitstream. I hope it will work better now.

closed 2024-07-12 by ufrisk