ufrisk/pcileech-fpga #117: issues with vivado building and other stuff

nascarZ, 2023-01-06:
Hello, so ive been trying to get started and im having issues with general building and trying to edit stuff
[image]
[image]
when trying to run the build
to just break down what i did:
opened vivado _> Window tab -> Tcl console -> I cd into the pciesquierrel folder on my desktop then i click the add sources on project manager -> add or create design sources -> src and ip folders i got in the pciesquierel folder from the zip -> and then run in console prompt vivado_generate_project.tcl -notrace and after a couple minutes just errors with some files...
And on top of that when i did try modifying the pcie_7x_0.xci like said in build.md it says the pcie_7x_0.xci is IP locked.
[image]
i tried searching and stuff but i couldnt really find anything.. any help?

nascarZ, 2023-01-06:
nvm im dumb

ufrisk, 2023-01-07:
I'm glad to see you managed to resolve the issue without any help. closing the issue.

closed 2023-01-07 by ufrisk