ufrisk/pcileech-fpga #119: Information about LeetDma

Remashes, 2023-01-29:
Hello, I would like to know how can I build a firmware for LeetDMA? Should I use Enigma, Squirrel or Screamer folder? and what device ID should I set?

Thank you for your help.

ufrisk, 2023-01-30:
Hi, the LeetDMA is directly compatible with the Squirrel. You can download the pre-built firmware and flash it, or download the Squirrel sources and modify them to your wishes.

I should add something about this in the documentation...

Relys, 2023-03-10:
So I just picked up a LeetDma card for 1299 Yen (~$185) from taobao. Very good deal. However, I'm having issues getting the jtag port to work. I plug in the cable and two USB To UART+JTAG interfaces show up by QinHeg Electronics USB\VID_1A86&PID_55DD&MI_00 & USB\VID_1A86&PID_55DD&MI_02 respectively. I've used Zadig to replace with WinUSB driver and modified the flash_screamer_squirrel.cfg for open ocd with the vid and pid. Maybe ftdi_layout_init is also different???

I've also tried Xilix Vivado Hardware Manager and nothing shows up. Any ideas?

ufrisk, 2023-03-12:
Many thanks for this info. I'm happy to see the issue is resolved.

closed 2023-03-12 by ufrisk

dvGrab, 2023-06-14:
It looks like LeetDMA devices are the same as NinjaDMA which are pretty affordable. Thanks for this, helped me out.

dexus1337, 2023-11-11:
@Relys first of all, thanks for your guide! Just what i was looking for. 

However, it doesnt seem to work for me, i get the following error message:

`F:\dma\flashing\bin>openocd.exe -f dump.cfg
Open On-Chip Debugger 0.12.0+dev (2023-07-24-11:41)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : CH347 Open Succ.
Info : clock speed 10000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Error: Unknown flash device (ID 0xffffff)

Info : Close the CH347. `

Also the board seems to hang after trying to run this command. The green (status?) led is constantly ON as well. Restetting the system will bring it back though.

Any chance you've run into similar issues?

z12crypto, 2024-01-17:
[image]
I have followed @Relys' method for flash compiling the firmware, but I encountered an error. What could be the possible issue?

Relys, 2024-01-24:
@dexus1337 @z12crypto Maybe try lowering the adapter speed

Relys, 2024-01-29:
> @Relys first of all, thanks for your guide! Just what i was looking for.
> 
> However, it doesnt seem to work for me, i get the following error message:
> 
> ```
> openocd.exe -f dump.cfg
> Open On-Chip Debugger 0.12.0+dev (2023-07-24-11:41)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
> Info : CH347 Open Succ.
> Info : clock speed 10000 kHz
> Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
> Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
> Error: Unknown flash device (ID 0xffffff)
> 
> Info : Close the CH347. 
> ```
> 
> Also the board seems to hang after trying to run this command. The green (status?) led is constantly ON as well. Restetting the system will bring it back though.
> 
> Any chance you've run into similar issues?

This is what happens when the FPGA  is not powered on but you're trying to connect via USB JTAG. Make sure the board is powered on.

Villageslayer, 2024-08-08:
Thanks for this info an Error prevented me from accidently flashing Enigma X1 FW Thanks again