ufrisk/pcileech-fpga #122: How to reboot the fpga without having to reboot the target machine

yaellem, 2023-04-05:
Hello, 

First of all thank you a lot for your incredible work !
I have a question, I am currently using the PCIE Screamer Squirrel and after being plugged for several hours on my target machine, I have 100% page failed for a probe or a dump, so I reboot my target and everything start working again. I would like to know if there is a way to reboot the fpga without having to reboot my target. 
Also I realized that the fpga datasheet on Lambda Concept was really lacking of infos about the switches. We do not know what is the meaning of the 3 switches, and the reset button especially that I thought was made to achieve what I want but when I press it, even if the LED blink like in a normal situation, I get the error : DEVICE: FPGA: ERROR: Unable to retrieve required Device PCIe ID [4,v4.10,0000] after launching a probe/dump.
So I have to reboot the target and everything is okay, but I am really lost on the switches meaning, and especially how to reboot or make the fpga working again without having to reboot the target machine (I also tried to unplug and re plug the fpga on the live target but it does the same that my other tries).

Thank you in advance :)

Here is the link of the lambda concept page with the datasheet at the bottom : http://docs.lambdaconcept.com/screamer/index.html

ufrisk, 2023-04-06:
Hold the reset button for 5-10s and then release it.

If you're lucky it will be detected properly, but PCIe hotplug is quite problematic. It may be that it's just not possible on your computer.

Doing it over Thunderbolt (which is hotplug) with an eGPU enclosure may be possible as well.

yaellem, 2023-04-07:
Thank you for you answer :)
Unfortunately after this operation, (the LED 1 is blinking but not each time), I get the error:
DEVICE: FPGA: ERROR: Unable to retrieve required Device PCIe ID [4,v4.10,0000] 

I tried different options like : Unplug -> look for change in material manually from the device manager (-> device disappear) -> Re plug -> Re check manually in device manager (-> device reappear). That works sometimes...
I am wondering why this situation happens with the board, is it from the OS that realized the device is not used for a while and deactivate it or something like that, is it coming from the board that is put in standby, then can not be detected anymore until a reboot where the bios will check for devices ? 
I tried changing in the target OS some alimentation setting, I do not know if that state is due to the OS or the board itself and if it is linked to some energy saving options.
Thank you for your advices, I'll tell later if my changes in the alimentation settings for PCIe worked.
I would have some questions too about your memprocfs API (which is great btw) but I'll do it on the appropriate repository.

Have a great day and thank you again for your answer :)

ufrisk, 2023-04-08:
Thanks for the update.

What you're describing closely resembles the wonders of PCIe hotplug on consumer hardware :( The PCIe ID [... 0000] means that the reset of the device went alright just that it haven't gotten picked up by the host system PCIe.

There isn't anything I'd be able to do about this so I'm closing the issue. If you have better success changing some settings pleae let me know though, it would be quite interesting for the future :)

closed 2023-04-08 by ufrisk

yaellem, 2023-04-21:
Hello,
I keep you update with the alternative ways I try to use to avoid this problem ^^ 
The settings I changed about the energy saving options of PCIE devices made that the board was not anymore put in standby by the OS, so as long as the target is turned on the board is not put in standby. However, when the target machine is itself in standby  (after long time of not using it - I could change that time setting too I think-), the board is put in standby too, which seems logical so far, but when the machine takes back its normal running state, the acquisition is not possible anymore. I mean that the pcie "ethernet" device is still recognize and shown in the device manager, but from the pcileech commandline tool I get a 100% page fail. And I cannot reset the board by any way because the machine is running and then the board would not be recognize anymore and i would get the error DEVICE: FPGA: ERROR: Unable to retrieve required Device PCIe ID [4,v4.10,0000].

So I try reloading the check of devices from the device manager, but the thing is that, it is recognized and shown in the devices list, and it's just the mem acquisition that is not possible anymore with the 100% page failed. The only solutions I found was either to reboot the target, either to change the host settings to never put the machine in standby.

I will keep you update if I discover something interesting new :)

Have a great day :)

ufrisk, 2023-04-21:
Thanks for the update, I guess one way would be to keep it busy as well, sending a TLP now and then, or maybe not?