ufrisk/pcileech-fpga #123: working on zynq？

Forever-APEX, 2023-04-07:
Hello,
First of all thank you a lot for your incredible work ! 
I am a student in china. so my english is not very good. As i know zynq have a linux system and a FPGA in a core. I wonder  could we do this in a pc?  linux read and write pc RAM. I am useing zu4ev working on it.
I just see you have some progress on zynq but have some problems.Can you tell me which ic you are working on it?
I have ask some zynq developer. They tell me is possible but need some times.As I have more information i will tell you.And DMA can run on PCIE 4x is possible?
But I'm curious how running DMA on Zynq is compatible with programs that run on 35T now？（translate by bing）or need redeveloped? 
This is my FPGA board.My email is qaz13521967875@gmail.com
[image]

ufrisk, 2023-04-08:
My plan a while back was to make PCILeech / MemProcFS run on the Zynq7000. I already have a AX7015 (XC7Z015) at hand.

PCILeech already runs on 32-bit ARM, albeit a bit slower and memory constraints is an issue for the more advanced forensic features, but not in general API usage. The Zynq 7000 is pretty much the Artix7 so the plan from my side was to just replace the USB/FT601 interconnect with the on-chip Zynq7000 AXI interconnect. The main benefit of this would be much lower latency.
 Otherwise there would be minimal changes to the PCILeech code required.

I ran into some minor issues, then the business case disappeared and then other priorities took over so this project is currently on hold from my side. I may look into it again in a few more months.

---

Your board with the Zynq Ultrascale (which is 64-bit and have much better memory bandwidth) is a much nicer fit for PCILeech/MemProcFS though. PCILeech/MemProcFS runs very well on 64-bit and this board have more memory - which makes it ideal.

It would require additional changes/redevelopments to PCILeech since the PCIe core is not directly compatible as far as I can see. I still believe it to be the better fit as compared to the Zynq7000.

This means additional work though, and the board is quite costly.

If I get the AX7015 to work I may look into this board as well.

ufrisk, 2024-01-17:
Unfortunately I have other plans right now. I don't forsee that I'll be able to look into the Zynq7000 series. Possibly Ultrascales down the road.

I'm closing this issue.

closed 2024-01-17 by ufrisk