ufrisk/pcileech-fpga #128: How to activate BusMaster ?

Youssix, 2023-06-16:
Hello ulfrisk,

Could you tell me which way should i look to enable BusMaster in my pcie config in vivado, i'm quite lost irght now 

Thanks.

ufrisk, 2023-06-16:
I don't think there is an option for that. It should get set at boot anyway. Also the xilinx pcie core disregards this value. You should be able to transmit even if it's unset.

If you mean bus master on upstream bridges you can't set that from your endpoint device.

Youssix, 2023-06-17:
What i mean is that when checking my device with lspci 

[image]

Mem- BusMaster-

But whenever i check the real device, it has Mem+ BUsMaster+

This is my question i can't find in the [pcileech_pcie_cfg_a7.sv](https://github.com/ufrisk/pcileech-fpga/blob/master/PCIeSquirrel/src/pcileech_pcie_cfg_a7.sv) where should i flip the bit ?

ufrisk, 2023-06-17:
As I already mention, this flag is not important for the Xilinx device. You can send/receive TLPs anyway even if it's against spec.

This value is usually set by BIOS as power-on/device enumeration. There are exception for some class codes and devices etc, but if you cloned a device it's likely that it's a driver that have unset this value.

It's not a setting in the core file as such, you'd have to perform a config space write which is some what complicated. It's possible to do this from the C side of things using the LeechCore library, or you could add something similar to the optional master abort reset I have here: https://github.com/ufrisk/pcileech-fpga/blob/6fdaa9a2c9d8680a25c86e6a9652e6cae3269567/PCIeSquirrel/src/pcileech_pcie_cfg_a7.sv#L354

If there is a need I guess I could look into adding automatic periodic re-enabling of this flag if it's been disabled.

Youssix, 2023-06-17:
Thank you for pointing this out,

It would be a good thing to have an option to do the periodic re enabling.

Most of the questions you got about cheating is because Vanguard anticheat disable busMaster of some pcie devices

ufrisk, 2023-06-17:
I'm not doing cheats. I know my tools are used for that, but it's not why I created them. Even if I don't really have issues with my tools being used for it I prefer not to get involved and stay away from it. My stance on these matters should be fairly well known by now.

In your case though re-enabling bus-master on the PCILeech device won't really do any good. Vanguard disable it on the upstream PCIe bridge/switch for downstream devices. The PCILeech device can't just unset the busmaster flag on the upstream devices from its view downstream. And as mentioned already the Xilinx PCIe core completely disregards this bit. It's possible to send/receive DMA request regardless of what this bit is set to on the actual device.

Still it would probably be a good idea for me to allow some additional flexibility in this area. But I'm afraid it won't really do any good in your use case. I can see some minor infosec uses for it though so it's probably worthwhile for me to look into it.

ufrisk, 2023-06-20:
I've added the ability to auto-set command register flags. By default it's disabled. If enabled by default it will set IO+, Mem+, BM+ at 1ms intervals.

Enabled it by changing this into 1: https://github.com/ufrisk/pcileech-fpga/blob/80c37f1eff50072b6de15db24e2444f830ba2666/PCIeSquirrel/src/pcileech_pcie_cfg_a7.sv#L209

To set other values than the defaults do it here: https://github.com/ufrisk/pcileech-fpga/blob/80c37f1eff50072b6de15db24e2444f830ba2666/PCIeSquirrel/src/pcileech_pcie_cfg_a7.sv#LL361C29-L361C29

A firmware rebuild is required.

I'm closing this issue since the enhancement suggestion is now implemented. I'm afraid it won't resolve your root issue though and there is not much downstream devices could do if BM is disabled on upstream bridges/switches.

closed 2023-06-20 by ufrisk