ufrisk/pcileech-fpga #129: Modifying CFG space

Haseeb18P, 2023-06-16:
Hey, is it possible to modify blocks 40, 60 & 70 in the cfg space? Would you be able to give me any instructions on how I can achieve this?

ufrisk, 2023-06-17:
I know people have been modifying the .xci file directly that generates the PCIe core. Also the generated core sources.

But I have not looked into this myself so I can't really help more than this in this matter.

Haseeb18P, 2023-06-17:
Would you happen to know the exact name of the .xci file that generates the PCIe core?

ufrisk, 2023-06-17:
https://github.com/ufrisk/pcileech-fpga/blob/master/PCIeSquirrel/ip/pcie_7x_0.xci

Haseeb18P, 2023-06-17:
After looking at the documentation on xilinx, it seems like that the MSI Control is stored at the 40 range, does this also seem like I'm looking in the right direction too you? Thanks in advance!
[image]

ufrisk, 2023-06-17:
Yeah, looks ok to me. This is a long and tedious process though, but best wishes with it 👍

closed 2023-06-17 by ufrisk