ufrisk/pcileech-fpga #130: Emulating any given device

Haseeb18P, 2023-06-18:
Hello do you have any instructions on how I can emulate a device, so the dma will act as the said device its configured as, so it can send and receive tlp packets?

ufrisk, 2023-06-18:
You don't need to emulate a device to send/receive TLPs. It works with the default bitstream.

If you wish to fully emulate a device it's not supported. Only some degree of config space emulation is supported.

I read a while ago about a Phd thesis from Cambridge University when they did full device emulation by running the firmware of a networking board inside a qemu emulator shuffling the TLPs effectively emulating the networking board in full. I don't seem to be able to find that thesis now though.

JRBusiness, 2024-01-26:
> You don't need to emulate a device to send/receive TLPs. It works with the default bitstream.
>
> If you wish to fully emulate a device it's not supported. Only some degree of config space emulation is [...]

Does that meant they running a stock firmware of a Network Board or they running a modified PCILeech firmware to emulate the Board in full?

ufrisk, 2024-01-30:
@JRBusiness the project was ThunderClap which is found here: https://github.com/thunderclap-io/thunderclap-qemu

It was way too complicated for my taste though and used some very high-end Intel FPGA hardware. For your uses I would think it to be mostly a dead end.

closed 2024-01-30 by ufrisk