ufrisk/pcileech-fpga #132: How to fully customize PCIe configuration space?

zr5177, 2023-07-06:
I want to customize all PCIe configuration spaces (including lines 40 and 60) with vivado. What should I do to achieve it? I can pay for it. This content is translated by machine, please forgive me.

ufrisk, 2023-07-07:
Hi,

This is a lot of work. You would have to edit the `pcileech_cfgspace.coe)` file, the `pcie_7x_0.xci` file and most probably also edit the verilog sources for the xilinx generated pcie core manually.

Unfortunately this is not something I'd be able to help with unfortunately. But I hope you'll have success in your project. I wish you the best with your DMA project.

BLK434, 2023-07-16:
If u still struggle on this issue, i'm glad to help.

Saran520, 2023-07-28:
> If u still struggle on this issue, i'm glad to help.

I need help

zr5177, 2023-08-03:
I need your help.

ufrisk, 2023-08-03:
I hope you all are able to help each others with this.

Unfortunately I'm not able to help with this since I haven't done it in the detail you require myself, also it's too close to games hacking which I don't wish to support (although I have no issues with the tools being used for it). I just don't wish to get involved.

Thanks for understanding and best wishes with these projects.

closed 2023-08-03 by ufrisk

focalos1337, 2024-03-02:
> I need help

I need help