ufrisk/pcileech-fpga #133: Base Address Register [AMD]

ekknod, 2023-07-07:
Hi. I have made own simple BAR support by using your shadow space as base.

it's something like this simple:
    wire            snoop_valid_rd = snoop_valid_rdwr & 
                                    ((snoop_data[31:25] == 7'b0000010) | (snoop_data[31:25] == 7'b0000000));          // Fmt[2:0]=000b (3 DW header, no data), CfgRd0/CfgRd1=0010xb
    wire            snoop_valid_wr = snoop_valid_rdwr & 
                                    ((snoop_data[31:25] == 7'b0100010) | (snoop_data[31:25] == 7'b0100000));          // Fmt[2:0]=010b (3 DW header, data), CfgWr0/CfgWr1=0010xb
                                    
Anyways. this is working fine with Intel machines, but not with AMD.
At first i was really sure it was actual this code fault, since its very lazy way but later tried actual well made BAR support,
and it had the same issue.

BAR support is required to communicate with device driver to properly mimic real device, this already works for Intel machines.
This is done just for researching purposes. I'm personally trying to find ways how cheaters can circumvent anti-cheats and
how anti-cheats can potentially catch them.

The issue:
Intel can read all addresses from BAR.
Amd can only read addresses (0x00, 0x80,0x100,0x180) (weird pattern)

ekknod, 2023-07-24:
Code is now publicly available. I did make own repository, hopefully you don't mind.
https://github.com/ekknod/pcileech-wifi/

- minimal BAR support
- wifi card "emulation"

ufrisk, 2023-07-24:
This is very nice, I'll take a good look once I'm back home from vacation (in a weeks time). I'm sure this will be useful for quite a few others 👍

ekknod, 2023-08-09:
I have added AMD support now wifi project. I decided to change approach for PIO module (provided by xilinx).
Hopefully my approach is okay. I changed your Multiplexer little bit to support PIO, hopefully i did it correctly.

https://github.com/ekknod/pcileech-wifi/blob/main/src/pcileech_pcie_tlp_a7.sv#L111

ufrisk, 2023-08-09:
This is some super awesome news 👍 Project completed then I guess?

Having full support (at least for smaller BARs) in the firmware opens up so many possibilities. And if you have some networking board emulation working it's super interesting for some research as well :)

I hope to finish up my slower API-based software implementation in the next few weeks, but it's quite boring to add support to all the different programming language APIs I provide. Also, I want to get some other changes in there as well...

ekknod, 2023-08-10:
"Project completed then I guess?"
That's correct. It was minimal project, what did little bit get out of hands since needed to do more than originally was planned to. classic scenario.

"I hope to finish up my slower API-based software implementation in the next few weeks, but it's quite boring to add support to all the different programming language APIs I provide. Also, I want to get some other changes in there as well..."
Ah.. I can feel the pain. That's indeed boring and easy to make errors.

closed 2023-08-10 by ekknod