ufrisk/pcileech-fpga #134: How to improve data port speed?

BLK434, 2023-07-16:
According to FT601 datasheet, it says this ic can run 5Gbps speed,convert to byte/s should be 625Mbyte/s, but my test say the speed is 180MByte/s. I'm trying to figure out how to reach 5Gbps speed, Does it determind by pcileech_ft601.sv? Or something else i didn't notice?

BLK434, 2023-07-16:
BTW I'm using USB3.2 GEN1 wire&USB3.2 GEN1 interface.

ufrisk, 2023-07-16:
FIFO side of this chip is 32-bits @ 100MHz, i.e. 400MB/s dynamically shared between rx/tx.

Protocol overhead in PCILeech (in the implemention I've got) is 1 dword per / 7 dwords of data transmitted. i.e. x 7/8

PCIe TLP overhead for larger rx is 12bytes per 128 bytes received, i.e. x 128 / (128 + 12).

400 * 7/8 * 128/(128+12) = 320MB/s theoretical when only bulk receiving data. Add to this that some time is required for transmitting data over this shared bus as well it will eat up max performance.

People on some really fast Intel computers have reported close to 250MB/s on optimal workloads, and this is very close to the theoretical max.

In reality 180MB/s isn't that bad. It's close to the practical max of the FT601 chip. If you can find a better chip please let me know.

BLK434, 2023-07-16:
Thanks for your quick answer! According your answer,could i get a conclusion that cable does effect the speed? Before u answer I just found some information from a board provider,they said use 3.2gen2 cable&interface could improve speed  to 240MByte/s.

ufrisk, 2023-07-16:
I don't think USB 3.2 will make a difference in itself since the FT601 is not capable of it. But a nice higher speed USB controller and a really fast CPU definitely makes some changes. And as you mention speeds around 240 MB/s have been possible, and it's very close to the theoretical maximum... so there isn't a whole lot more I can do as things stand right now...

BLK434, 2023-07-17:
Ok thanks i'll keep research this part!

closed 2023-07-17 by BLK434