ufrisk/pcileech-fpga #135: [Squirrel] Flashing but nothing appears in device manager and can't pcieleech

Youssix, 2023-08-09:
Hey,

I have an issue with my squirrel dma card.

I am flashing it but it's not working at all, i see nothing in the device manager ?

The led1 is blinking , led 2 off and led 3 ON.

I did flash it , 

when i run : 

PS D:\DMA\Pcieleech> .\pcileech.exe probe -v -device fpga://devreload=1

DEVICE: FPGA: ERROR: Unable to retrieve required Device PCIe ID [4,v4.11,0000]
PCILEECH: Failed to connect to the device.

My middle led blinks then stop.

ufrisk, 2023-08-09:
The flashing is successful. It's the expected behavior.

You however have a bad PCIe connection for some reason. Try shutting down the PC and then power it up again to see if things have improved.

Youssix, 2023-08-09:
Are you suggsesting to try another PCIe port ? 

i'll try again then but it was working fine until i try to flash recently with a custom pcileech wifi fw, even when flashing to my old firmware i was using it's still not found in device manager.

the flashing gets right i believe then if you say so , what could be the problem , i already tried many time to reboot the computer already.

ufrisk, 2023-08-10:
Sometimes the PCIe is not working if you mess up your custom config space enough. The flash seems however to be OK.

You can try to flash the default firmware I provide here on Github to see if it works better and start from there.

Let me know if it works better with the default firmware.

Youssix, 2023-10-31:
Hello @ufrisk ,

So i just tried it today and i have this result : 

PS D:\DMA\Pcieleech> .\pcileech.exe probe -v -device fpga://devreload=1

DEVICE: FPGA: ERROR: Unable to retrieve required Device PCIe ID [4,v4.12,0000]
PCILEECH: Failed to connect to the device.

Same error, i even tried to switch the pcie lane, nothing helps.

What could be the problem ? the card is dead ?

Youssix, 2023-10-31:
PS D:\DMA\MemProcFS_files_and_binaries_v5.8.15-win_x64-20231022> .\MemProcFS.exe -v -vv -device fpga

----- FPGA DEVICE CONFIG REGISTERS: CORE-READ-ONLY     SIZE: 40 BYTES -----
0000    89 ab 00 00 28 00 00 00  04 0c 04 00 00 00 00 00   ....(...........
0010    d8 ac b6 60 05 00 00 00  da ac b6 60 05 00 00 00   ...`.......`....
0020    00 00 02 00 ff ff ff ff                            ........

----- FPGA DEVICE CONFIG REGISTERS: CORE-READ-WRITE    SIZE: 30 BYTES -----
0000    cd ef 04 00 1e 00 00 00  a0 86 01 00 00 00 00 00   ................
0010    ee 10 07 00 ee 10 66 06  02 3c 00 00 00 00         ......f..<....

----- PCIe CORE Dynamic Reconfiguration Port (DRP)  SIZE: 0x100 BYTES -----
0000    00 00 00 01 00 02 00 00  00 00 00 00 00 00 f0 00   ................
0010    ff ff 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0020    00 00 00 00 00 00 00 00  00 00 00 40 00 00 00 00   ...........@....
0030    00 00 09 02 30 00 22 7f  02 00 01 00 00 03 11 0c   ....0." ........
0040    02 a1 00 43 01 00 1f fd  7f ff 09 ff 01 20 01 48   ...C.... .... .H
0050    00 05 01 60 11 9c 00 00  00 00 00 00 00 00 00 00   ...`............
0060    00 00 10 60 00 02 40 21  00 40 3d 48 00 23 00 00   ...`..@!.@=H.#..
0070    00 00 00 00 00 00 00 00  00 00 00 00 00 15 00 01   ................
0080    00 01 00 00 00 01 00 00  00 01 00 00 00 01 00 00   ................
0090    00 01 00 00 00 01 00 00  00 00 00 00 00 00 00 00   ................
00a0    00 00 00 00 00 00 00 00  00 02 00 00 12 34 10 18   .............4..
00b0    00 0b 00 01 00 11 00 00  00 00 00 00 00 01 00 00   ................
00c0    00 28 00 41 ff ff ff ff  00 e0 00 00 80 08 00 22   .(.A..........."
00d0    07 ff 03 52 02 48 00 08  00 40 0e 84 fa ac 00 00   ...R.H...@......
00e0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00f0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................

----- PCIe SHADOW CONFIGURATION SPACE (only user set values) SIZE: 0x1000 BYTES -----
0000    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0010    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0020    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0030    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0040    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0050    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0060    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0070    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0080    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0090    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00a0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00b0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00c0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00d0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00e0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00f0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0100    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0110    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0120    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0130    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0140    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0150    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0160    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0170    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0180    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0190    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01a0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01b0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01c0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01d0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01e0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01f0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0200    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0210    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0220    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0230    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0240    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0250    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0260    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0270    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
[... 194 lines cut ...]
0ea0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0eb0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0ec0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0ed0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0ee0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
[... cut ...]
0fb0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0fc0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0fd0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0fe0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0ff0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................

DEVICE: FPGA: ERROR: Unable to retrieve required Device PCIe ID [4,v4.12,0000]
MemProcFS: Failed to connect to memory acquisition device.

Youssix, 2023-10-31:
[image]

Youssix, 2023-10-31:
It seems like a lot of people have this problem no ?

Youssix, 2023-10-31:
The pcie slot is showed at empty by my motherboard at bios, is it normal ?

Youssix, 2023-11-13:
Hardware issue, ordered a new one and it works thanks.

ufrisk, 2023-11-13:
Thanks for the update, I'm glad to see it works now. Best wishes with your DMA attacking.

closed 2023-11-13 by ufrisk