ufrisk/pcileech-fpga #139: why the firmware speed will be lower if i make all config space from .coe file?

xingkong158, 2023-08-21:
I make all config space from .coe file, but i will get lower read memory speed, how can i get the speed back?

ufrisk, 2023-08-22:
Hi,

How does the config space when doing `lspci -d 10ee:0666 -xxxx` (in Linux as root) look like on your modified device as compared to the correct one you're trying to clone? (replace 10ee:0666 here with whatever vendor/device id you're using).

Chances are that you're either looking at a very basic legacy device or that there are some errors in the config. Some computers disallow reads using the "normal" algorithm with 4kB read / PCIe tag on old legacy hardware or when there is some errors (I think related to capabilities, but I'm not completely sure).

xingkong158, 2023-08-23:
ok thanks brother, but i see [image: image.png]here, if i use our source
code, this number how to custom it?

ufrisk, 2023-08-23:
Hi, I don't see the image you're trying to show. But if you do `lspci -tv` you'll see all the devices and their device ids in a tree structure (or you may have to do `lspci -n` for that I don't remember.

xingkong158, 2023-08-24:
[image]
brother this one, and i want to know where can custom all configuration space header?

KernelKrusha, 2023-08-24:
@xingkong158 what software is that?

ufrisk, 2023-08-24:
@xingkong158 You may try to edit the [pcie xci file](https://github.com/ufrisk/pcileech-fpga/blob/master/PCIeSquirrel/ip/pcie_7x_0.xci) before generating the core. If this setting is unavailable to customize to its fullest extent there I know others what have then edited the auto-generated verilog files that Vivado generates for the PCIe core.

xingkong158, 2023-08-25:
> @xingkong158 what software is that?

telescan

xingkong158, 2023-08-27:
[image]
can we change Base Address Register 0 from source code?

xingkong158, 2023-08-27:
[image]
i want to change it to same, but if i change here
[image]
i cant get currect value. 

the picture currect value = Captured Slot Power Limit Value 01001011, 01001011 ="4B". but i change "SLOT_CAP_SLOT_POWER_LIMIT_VALUE" to 4B, i still cant get right value with my firmware.

ufrisk, 2023-08-27:
Hi,

Editing the core manually is not something I'm experienced in and I'd be of very limited help asking questions about that. I just haven't looked into it at the detail level you require :\

But,

There is something called DRP, which is a debug port that allows you from software to set some of the internal workings of the core before it's brought online.

I haven't tested this, but  you can try to replace the 4 lines of `64'h00000000_00000000,` [here](https://github.com/ufrisk/pcileech-fpga/blob/80c37f1eff50072b6de15db24e2444f830ba2666/PCIeSquirrel/src/pcileech_com.sv#L70) with:
```
            64'h00000000_00000000,
            64'h5100ffff_801c2377,  // DRP ADDR: SLOT_CAP_SLOT_POWER_LIMIT_SCALE/VALUE
            64'h120cffff_801a2377,  // DRP DATA: LIMIT_SCALE=0,LIMIT_VALUE=4B,SSL_MESSAGE_AUTO=0
            64'h20002000_80022377,  // DRP WRITE EN
```

to write to the DRP debug port before the PCIe core is brought online. Maybe this will help (or maybe not).

More information about the DRP interface is found in the Xilinx PCIe core guide / data sheet.

Also, you may edit the file [pcie_7x_0.xci](https://github.com/ufrisk/pcileech-fpga/blob/master/PCIeSquirrel/ip/pcie_7x_0.xci) before generating the core. I suspect you've already tried this though and that it didn't work.

Please let me know how it goes with using the DRP debug port, and best wishes.