ufrisk/pcileech-fpga #141: M.2 NVMe M-key to PCIe adapter issues

IgnPilv, 2023-08-30:
Hello, I am running Lenovo Legion 5, which has a M.2 slot. 

I have a DMA card (tested in a PCIe slot on another machine, works well)

I used [this adapter](https://www.amazon.com/GLOTRENDS-Extension-Cable-Firewire-UEX108/dp/B09ZSP7HFR/ref=sr_1_11?keywords=m.2+to+pcie+adapter&sr=8-11) to connect it to the laptop.

pcileech.exe recognizes the device, gets vendor id, device id, firmware version, however fails testmemread at 0xff0. I believe the fault is in the adapter, however I didn't find any info on this, so figured I'd share for other people as well.

Is compatability for DMA boards possible via adapters? If so, what are the specs necessary for this?
Perhaps [this](https://www.delock.com/produkt/64131/merkmale.html) one would be fine?

ufrisk, 2023-09-01:
Possibly, I'm really hesitant in giving advice around this. I've had people demanding money from me when their purchase didn't work in the past. But best wishes with your DMA attacks 👍

IgnPilv, 2023-09-07:
Understandable, thank you.

closed 2023-09-07 by IgnPilv