ufrisk/pcileech-fpga #142: hi, how can i change here from config space in core_top file?

xingkong158, 2023-09-01:
[image]
hi, Ulf
i dont know why, i cant find somewhere to let it be changed, can you tell me where can change it?

ufrisk, 2023-09-02:
Hi,

I'm sorry, but I actually haven't looked into extreme customizations of the PCIe core on this level. For the use cases I had it's been sufficient to do simpler customizations.

Some ideas though. If these values exist in the "DRP" debug port you may be able to override these values before the PCIe core is brought online by the PCILeech firmware.

Also, others have edited the PCIe core verilog source code files to achieve a pretty much full customization, so it's certainly possible. I haven't done this though.

Sorry for not being able to give a better answer than this right now :\

xingkong158, 2023-12-22:
hi brother, i just want to ask you how to do this config space?
[image: image.png]
it is different normal devices config space right?

ufrisk, 2023-12-22:
I have not looked into changing those specific bytes, but as I mentioned it's probably possible to do that by using the "DRP" debug port or by editing the PCIe core either the verilog files or the .xci file.

You could look into this setting and similar ones to see if it's of any help: https://github.com/ufrisk/pcileech-fpga/blob/e13b8378e6d2d271a70b5e5c687e8e52fa368f09/PCIeSquirrel/ip/pcie_7x_0.xci#L428C119-L428C119

closed 2024-01-17 by ufrisk