ufrisk/pcileech-fpga #143: Why Tiny alog? How to fix this

BLK434, 2023-09-07:
Hello ufrisk,I'm able to modify about 80% cfgspace, But i found same frimware got differrent result about R/W speed,some pc output"Tiny alog auto select" and  speed decrease ,But other PC run full speed,this is so confuse me,I'm not professional about windows,could you give me some information?

zr5177, 2023-09-07:
Yes, I have these confusions too. How to solve Tiny tlp? Or what causes it.

ufrisk, 2023-09-09:
Some hardware is more sensitive to legacy hardware or broken config spaces and won't allow you to perform 4KB PCIe reads - hence PCILeech downgrades to "tiny algorithm" automatically in those cases. I don't really know why it works fine on some hardware and not on other hardware.

Normal algorithm is 4kB PCIe reads.
Tiny algorithm is 128 byte PCIe reads.

Issue with Tiny algorithm and PCILeech is that PCILeech will quickly exhaust available PCIe tags due to the relatively high USB latency and this will result in a rather large performance impact as compared to the normal 4kB PCIe reads. If a 4kB normal PCIe reads will use 1 tag a tiny read of 4kB will use 32 tags (32*128 = 4kB).

BLK434, 2023-09-13:
So this issue depends on user motherboard ？Does there have any way to solve

ufrisk, 2023-09-13:
It depends on the motherboard (or even BIOS version) together with the firmware. Sometimes flashing the default firmware will resolve things and then there is an error with the modified firmware. Sometimes this does not work and if a BIOS flash/upgrade does not work I know some have purchased another motherboard.

My best guess is that this is due to a bad interaction with custom firmware, it's quite common from what I've heard...

closed 2024-01-17 by ufrisk