ufrisk/pcileech-fpga #145: Beginner's Inquiry: Unexpected FPGA config values when reading from even addresses

sadmess, 2023-10-27:
My device is not EnigmaX1, but its chip is xc7a75t and the storage is mt25ql256. I tried to compile the firmware of EnigmaX1 and flash it in.

After rebooting, the device was recognized by the host (a wireless network card of 0x0666), but some errors occurred when reading with leechcore communication.

The code from leechcore is as follows:

`
BOOL DeviceFPGA_GetDeviceID_FpgaVersionV4(_In_ PDEVICE_CONTEXT_FPGA ctx)
{
    WORD wbsDeviceId, wMagicPCIe;
    DWORD dwInactivityTimer = 0x000186a0;       // set inactivity timer to 1ms ( 0x0186a0 * 100MHz ) [only later activated on UDP bitstreams]
    if(!DeviceFPGA_ConfigRead(ctx, 0x0008, (PBYTE)&ctx->wFpgaVersionMajor, 1, FPGA_REG_CORE | FPGA_REG_READONLY) || ctx->wFpgaVersionMajor < 4) { return FALSE; }
    
    DeviceFPGA_ConfigRead(ctx, 0x0009, (PBYTE)&ctx->wFpgaVersionMinor, 1, FPGA_REG_CORE | FPGA_REG_READONLY);
    DeviceFPGA_ConfigRead(ctx, 0x000a, (PBYTE)&ctx->wFpgaID, 1, FPGA_REG_CORE | FPGA_REG_READONLY);
    printf("ctx->wFpgaID %x\n", ctx->wFpgaID);
    ...
    DeviceFPGA_ConfigWrite(ctx, 0x0008, (PBYTE)&dwInactivityTimer, 4, FPGA_REG_CORE | FPGA_REG_READWRITE);
    // PCIe
    DeviceFPGA_ConfigRead(ctx, 0x0008, (PBYTE)&wbsDeviceId, 2, FPGA_REG_PCIE | FPGA_REG_READONLY);
    printf("wbsDeviceId %d\n", wbsDeviceId);
}
`
I added printf, the print result of ctx->wFpgaID is 1, and the expected value should be 9. In addition, I also tried to modify the firmware's

`
parameter       PARAM_DEVICE_ID = 9,
`
to
`
parameter       PARAM_DEVICE_ID = 10,
`
Then the print result of ctx->wFpgaID is 2, and the expected value should be 10.

I also found that the print result of wbsDeviceId is not normal. The device's pci bus is 13 and the device is 0 on my machine, but the print result of wbsDeviceId is 0x500. The expected value should be 0xD00.

From the above experiments, I found that when reading bytes at even addresses, the binary of 9 is 1001 but it becomes 0001, the binary of 10 is 1010 but it becomes 0010, and the binary of 13 is 1101 but it becomes 0101.

I am a beginner in FPGA and have not read a lot of related materials. If my question is naive, please give me relevant documents to study. Thanks in advance.

sadmess, 2023-10-28:
```
# OLD  ft601_data[19]
set_property PACKAGE_PIN T20  [get_ports ft601_data[19]]
# NEW  ft601_data[19]
#set_property PACKAGE_PIN T18  [get_ports ft601_data[19]]
```
Make the following changes and that's it, everything works now

ufrisk, 2023-10-28:
Good to see you managed to resolve the issue.

If I remember correctly this was the change that was made from the prototype enigma version (which should be very rare) to the later enigma version. Nice to see it's working for you :)

But my project does not support the prototype enigma version (and like I mentioned it should be extremely rare) so I'm closing this issue.

Best wishes with your DMA attacks 👍

closed 2023-10-28 by ufrisk