ufrisk/pcileech-fpga #146: Failed reading a memory display issue

kirysnik, 2023-11-22:
Hello,
I was using the pcie squirrel for a while and there was no problem at all, but yesterday something happened to my current board. Device is connected but I can't read the memory from the target computer. I already tried different firmware, flashed even back to the default firmware but it still the same. Of course IOMMU disabled.

[image]

ufrisk, 2023-11-22:
You should try this after a fresh boot after you unlocked the computer and logged on to Windows. Also have VT-d / IOMMU disabled in BIOS (as  you mention). If you use the Thunderbolt to access DMA you'd need to disable that security level (set to disable or legacy) also.

for even additional details you may use: `pcileech.exe display -min 0x1000 -device fpga -v -vvv`

also, you may also try (once again the first command after a fresh boot: `pcileech.exe display -min 0x1000 -device fpga://algo=1 -v -vvv`

Please let me know how it goes.

closed 2024-01-17 by ufrisk

tchernobogdev, 2024-03-14:
Hello, I am getting a similar issue on the Squirrel board as well. When running probe or dump I get 100% page failure and running any test commands like `sudo ./pcileech testmemreadwrite -device fpga -v` or `sudo ./pcileech display -min 0x1000 -device fpga -v -vvv` fail. 

LED 3 is green on the board, cpu virtualization and iommu are disabled on target and the Squirrel board is plugged into a pcie slot. Have also tried fresh booting in a variety of ways, reinstalled dependencies etc etc.

My control computer is a pi4 running the aarch64 latest release binaries. Target computer has AMD cpu and is running windows 10. Are there any other anti-DMA settings I am missing besides iommu, thunderbolt (N/A here) and cpu virtualization, or what might be the issue?