ufrisk/pcileech-fpga #147: Does LeetDMA Pro V2 use the enigma-x1 dictionary code?

lyciumlee, 2023-11-28:


lyciumlee, 2023-11-28:
I have compiled the enigma-x1 srcs. I use the Vivado 2020.2 to generate the bit stream. But I do not know how to flash it to the device. My computer is a Windows laptop. In device manager, I can not find my leetdma v2 Pro. All type-C ports of the leetdma video can not detect the device. I try my best to find the drivers of the bord. but I can not find it.

lyciumlee, 2023-11-29:
I will try. Thanks!

closed 2024-01-17 by ufrisk

tisoz, 2024-07-01:
> I have compiled the enigma-x1 srcs. I use the Vivado 2020.2 to generate the bit stream. But I do not know how to flash it to the device. My computer is a Windows laptop. In device manager, I can not find my leetdma v2 Pro. All type-C ports of the leetdma video can not detect the device. I try my best to find the drivers of the bord. but I can not find it.我已经编译了 enigma-x1 源代码。我使用 Vivado 2020.2 生成比特流。但我不知道如何将其刷入设备。我的电脑是 Windows 笔记本电脑。在设备管理器中，我找不到我的 leetdma v2 Pro。leetdma 视频的所有 Type-C 端口都无法检测到设备。我尽力找到了 bord 的驱动程序。但我找不到它。

hi , where i can find enigma-x1 srcs . thansk bro