ufrisk/pcileech-fpga #149: about memory issue

CaS1ow, 2023-12-07:
hi i am using 16+16gb ram
name: g.skill ddr4 32g pc4-28800 cl16 trident z royal c

i have a problem with this ram

dma test not work i tested alot of fw and custom fw and i changed asus mother board to msi motherboard but same issue

checked bios vt,iommu,fast,secure boot off and checked turned off xmp

here is my error
1.
DEVICE: FPGA: ERROR: Unable to retrieve required Device PCIe ID [3,v4.9,0000]
MemProcFS: Failed to connect to memory acquisition device.

2.

[CORE]     Initialization Failed. Unable to locate valid DTB. #2
VmmProc: Unable to auto-identify operating system for PROC file system mount.
         Specify PageDirectoryBase (DTB/CR3) in -cr3 option if value if known.
[CORE]     Failed to initialize.

i always do powercycle
this errors only showing when i install gskill ram

if i install 2nd pc's samsung ram, dma test works fine

CaS1ow, 2023-12-07:
please give me some advice
setting bios or just buy new ram will clean my brain..

finally i found how to fix that error but i wanna save my money i thought error's problem was my asus motherboard compatibility so i bought msi but finally found the problem is ram

CaS1ow, 2023-12-07:
i am using immortaltech.net warrior dma

CaS1ow, 2023-12-07:
i think i fixed it changed bios
1.memory fast boot disabled
2.security device support disabled

CaS1ow, 2023-12-07:
ok another problem appear
if i turn on xmp
```
FPGA: TINY PCIe TLP algrithm auto-selected!
LcMemMap_AddRange: 0000000000000000-000000000009ffff -> 0000000000000000
LcMemMap_AddRange: 0000000000100000-00000000ffffffff -> 0000000000100000
LeechCore v2.14.2: Open Device: fpga
[CORE]     Initialization Failed. Unable to locate valid DTB. #2
VmmProc: Unable to auto-identify operating system for PROC file system mount.
         Specify PageDirectoryBase (DTB/CR3) in -cr3 option if value if known.
[CORE]     Failed to initialize.

[CORE]     SHUTDOWN COMPLETED (0000017BB44D8040).
```
after fresh reboot
```
DEVICE: FPGA: ERROR: Unable to retrieve required Device PCIe ID [3,v4.9,0000]
MemProcFS: Failed to connect to memory acquisition device.
```

ufrisk, 2023-12-07:
I'm not really able to give support on all those 3rd party devices with custom firmware that is sold. They don't contribute anything to my project financially and they just leave me with the support tab unfortunately. Also the custom firmware is having great many unknowns which makes it very hard for me to support.

It's how it is, my projects are open source so it's fine to do this I guess, but I'm not too happy giving out free support out of my time just so that others can earn a lot of money.

Version 4.9 is pretty old though, it's version 4.12 now and there have been a few bug fixes since then.

If it works without XMP can't you just run it that way? The performance gain of XMP isn't massive anyway so just disable it when doing DMA and enable it other wise?

closed 2024-01-17 by ufrisk