ufrisk/pcileech-fpga #152: Xilinx PCIe parameters

andreiw, 2023-12-15:
Poking around the generated files I saw UR_INV_REQ, UR_CFG1  and UR_ATOMIC... all set to true. Maybe these, when set to false, would allow these TLP kinds to not be handled automatically by the core? Have you tried messing with these?

How is the PCIe IP implemented? Is that a hard block on the FPGA? Or are the some generated sources that I'm just not seeing in the build?

ufrisk, 2023-12-16:
Xilinx 7series have a Hard-IP on-silicon PCIe block so it wouldn't be in the sources except for the auto-generated stub core sources.

closed 2023-12-16 by ufrisk

andreiw, 2023-12-18:
So what you're saying is that the UR_CFG1 and the like are hard coded and are not configurable?

ufrisk, 2023-12-19:
That is not what I said. You asked how the PCIe IP was implemented and I answered it was a hard-ip block.

For additional information about the Xilinx PCIe IP core and its workings I can recommend the documentation at: https://docs.xilinx.com/v/u/en-US/ug477_7Series_IntBlock_PCIe