ufrisk/pcileech-fpga #153: Q on receiving data from FPGA

andreiw, 2023-12-16:
Say I'm  reading an FPGA register (e.g. FPGA_REG_CORE | FPGA_REG_READWRITE)... how would I distinguish a reply to a read from a TLP coming in? Both have the 0xEzzzzzzz header, after all. Or is this inherently racy?

andreiw, 2023-12-16:
A TLP fragment will have  a status field like 0xeffff400 or 0xe000040.

A read from FPGA_REG_CORE will have a status like 0xeffffff3. So that _3_ works, I suppose. And _1_ signals for a FPGA_REG_PCIE. Am I on the right path? But what prevents replies from mixing with TLP receiving?

ufrisk, 2023-12-16:
The LeechCore protocol is 8 DWORDs long. 1 status DWORD and 7 data DWORDS.
the 4 hi-bits are set to 'E'
then there are 4 status bits for each data DWORD. The two high-bits are context dependent, the low two bits tell which type of data there is:
00 = TLP
01 = PCIe CFG
10 = Loopback
11 = Internal CFG

I hope this explains it a bit, and best wishes with your project!

closed 2023-12-16 by ufrisk

andreiw, 2023-12-18:
Thanks a lot!