ufrisk/pcileech-fpga #158: pls help me

faxvs, 2024-02-22:
When I mimic a simulated device

Squirrel's source code does not compile properly in vavido

But other 75t core source code can be compiled normally.
This is an error:

[DRC UTLZ-1] Resource utilization: RAMB18 and RAMB36/FIFO over-utilized in Top Level Design (This design requires more  RAMB18 and RAMB36/FIFO cells than are available in the target device.This design requires 109 of such cell types but only 100 compatible sites are available in the target device. Please  analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected.If so, please consider targeting a larger device.) im using 35T 。 i want know How @ufrisk @ekknod

faxvs, 2024-02-22:
@ufrisk 请你帮帮我

ufrisk, 2024-02-22:
Hi, please understand that this is an open source project. As such I'm not really paid to do this except the quite few kind sponsorships I get.

I'll happily help out in fixing errors in the project code should you find any.

I unfortunately do not have time to give general free help/advice in how to code in verilog and how to use vivado. The error message says it's over-utilized. The 35T is a smaller chip and have less resources than the 75T. Your design seems to be too big to fit properly on the 35T. Maybe you can find ways to shrink it or maybe it's not possible. I don't know.

I'm however closing this issue since it's not really an issue with my software in this repo. Best wishes with your project.

closed 2024-02-22 by ufrisk