ufrisk/pcileech-fpga #159: About solutions to some problems and some overlooked code.

zr5177, 2024-02-24:
Some questions about PCie configuration space customization
Use COE to highly customize everything except VEND_ID and DEV_ID: set rw[203]<= 1'b0; // CFGTLP ZERO DATA parameter in pcileech_fifo.sv to 0, if you want to use the configuration space defined by COE It is writable, please set CFGTLP PCIE WRITE ENABLE to 1. Then set the EXT_CFG_CAP_PTR and EXT_CFG_XP_CAP_PTR parameters in pcie_7x_0_core_top.v to 01. 01 means that you want the coe coverage to start from the 0x04 address of the configuration space. If the DEV_CAP_MAX_PAYLOAD_SUPPORTED parameter in your COE file is not 0, please turn on the MPS_FORCE parameter, search for the cfg_force_mps parameter, and assign it the value of DEV_CAP_MAX_PAYLOAD_SUPPORTED indicated by your coe to solve the minor TLP problem.
Bit 13 of the status register is set to 1 after active communication. This problem can be solved by setting rw[20] <= 1; in pcileech_pcie_cfg_a7.sv to 1. But it is not recommended to set rw[21] <= 0; to 1, because the value of the command register is usually controlled by the hardware driver.
I apologize for initiating this question which is not a question, this question is to share some issues and knowledge that I encountered and solved. If others have any questions, you can discuss them together. Finally, I wish ufrisk a happy New Year (Chinese New Year) (*^_^*)

focalos1337, 2024-02-28:
sorry to bother you, "Use COE to highly customize everything" refer to the pcileech_cfgspace.coe?

zr5177, 2024-02-28:
Yes, the bram_pcie_cfgspace IP core's pcileech_cfgspace.coe file can fully customize all PCIe configuration spaces given a specific PCIe IP core configuration.

focalos1337, 2024-02-28:
> Yes, the bram_pcie_cfgspace IP core's pcileech_cfgspace.coe file can fully customize all PCIe configuration spaces given a specific PCIe IP core configuration.

thank you very much.and some questions
#"Then set the EXT_CFG_CAP_PTR and EXT_CFG_XP_CAP_PTR parameters in pcie_7x_0_core_top.v to 01. 01 means that you want the coe coverage to start from the 0x04 address of the configuration space."
why set 01 rather 04
like this
```
parameter [5:0]   EXT_CFG_CAP_PTR = 6'h04,    
parameter [9:0]   EXT_CFG_XP_CAP_PTR = 10'h04,
```

zr5177, 2024-02-28:
In this parameter, each increment of one digit represents a 64-bit extension of the coe coverage, which corresponds to 4 addresses in the configuration space. Therefore, the parameter 01 represents coverage starting from address 0x04, while the parameter 04 represents coverage starting from address 0x10 (in hexadecimal).

CottonCollecter, 2024-02-28:
epic post

focalos1337, 2024-02-29:
> In this parameter, each increment of one digit represents a 64-bit extension of the coe coverage, which corresponds to 4 addresses in the configuration space. Therefore, the parameter 01 represents [...]

ok，thank you very much , i think i figure it out

focalos1337, 2024-02-29:
so sorry to bother you again, if DEV_CAP_MAX_PAYLOAD_SUPPORTED =2, what value should cfg_force_mps be set to?
`    .cfg_force_mps                              ( 3'b010 ),
`
is correct?

focalos1337, 2024-02-29:
I tried according to the content of the issue and it seems that I didn't get the correct configuration space and only a few values ​​were changed. Here are my steps

1.dump the network card info with telescan，and converted to dowrds
2.copy the dwords to pcileech_cfgspace.coe 
3.set rw[203]<= 1'b0 in pcileech_fifo.sv to 0 and set CFGTLP PCIE WRITE ENABLE to 1.
4.set rw[20] <= 1; in pcileech_pcie_cfg_a7.sv 
The above steps are all done using vscode
5.open vivado, open tcl console ,cd C:/Users/admin/Desktop/Dev/pcileech-fpga-master/PCIeSquirrel, source vivado_generate_project.tcl -notrace
6. navigate to the "sources" box and navigate as such pcileech_squirrel_top > i_pcileech_pcie_a7 : pcileech_pcie_a7 then double click on the file with the yellow square labelled i_pcie_7x_0 : pcie_7x_0.
7. just modify the vendor id and device id
8.cllick Generate

9.using vscode,search EXT_CFG_CAP_PTR EXT_CFG_XP_CAP_PTR and so on ,set to the following value
  parameter [5:0]  EXT_CFG_CAP_PTR = 6'h01,
  parameter [9:0]  EXT_CFG_XP_CAP_PTR = 10'h01,
  parameter         MPS_FORCE = "TRUE",
  .cfg_force_mps                              ( 3'b010 ),
10,back to vivado,click generate bitstream

Are there any steps I'm missing? Below is my lspci information, it only changed very few values
`00: ec 10 68 81 07 00 10 00 02 00 00 02 10 00 00 00
10: 00 00 90 fc 00 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 ee 10 07 00
30: 00 00 00 00 40 00 00 00 00 00 00 00 2d 01 00 00
40: 01 48 03 78 0b 00 00 00 05 60 80 00 00 00 00 00
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
60: 10 00 02 00 e2 8f 68 00 10 28 00 00 12 f4 03 00
70: 40 00 12 10 00 00 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00
90: 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
a0: 00 00 00 00 00 00 00 00 ff ff ff ff ff ff ff ff
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff`

zr5177, 2024-02-29:
您的修改方式，貌似跟我经常使用的不太一样，我在上面强调的是在VIVADO中直接修改IP核内的pcie_7x_0_core_top.v文件。而根据你的配置空间表现来看，貌似有几个问题：pcie IP核的EXT_CFG_CAP_PTR参数的值是 2A, bram_pcie_cfgspace IP 核没有正确识别到COE的更改，可能是因为没有重新打开IP核保存修改的COE文件。

zr5177, 2024-02-29:
Your modification method seems different from what I often use. I emphasized above that I directly modify the pcie_7x_0_core_top.v file within the IP core in VIVADO. Based on the performance of your configuration space, there seem to be some issues: the value of the EXT_CFG_CAP_PTR parameter of the PCIe IP core is 2A, and the bram_pcie_cfgspace IP core has not correctly recognized the changes to the COE, possibly because the modified COE file wasn't reopened and saved in the IP core.

focalos1337, 2024-03-01:
So I only need to execute set_property is_management false [get_files pcie_7x_0.xci] in step nine
Then modify the pcie_7x_0_core_top.v file in vivado, right?

kilmu-pro, 2024-03-02:
sorry to bother you
Do you know the possible reasons why the main computer cannot be turned on and remains in a black screen interface after flashing a new firmware in this way?

kilmu-pro, 2024-03-02:
> sorry to bother you Do you know the possible reasons why the main computer cannot be turned on and remains in a black screen interface after flashing a new firmware in this way?

I replaced a card and now it boots, but I see no big change in the config space

> 00: 86 80 23 27 00 00 10 00 02 00 00 02 10 00 00 00
10: 00 00 90 fc 00 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 ee 10 07 00
30: 00 00 00 00 40 00 00 00 00 00 00 00 ff 01 00 00
40: 01 48 03 78 08 00 00 00 05 60 80 00 00 00 00 00
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
60: 10 00 02 00 e2 8f 68 00 10 28 00 00 12 f4 03 00
70: 40 00 12 10 00 00 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00
90: 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
c0: 00 00 00 00 00 00 00 00 c8 23 d0 01 0d 00 00 08
d0: 00 80 40 05 00 00 00 00 00 00 00 00 00 00 00 00
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

zr5177, 2024-03-02:
It seems many people are facing the same issue. I will describe in more detail how to modify the corresponding parameters here. First, launch Vivado and open your PCILeech-FPGA project, ensuring the "CFGTLP ZERO DATA" parameter is set to 0. Open the "PCIE_7X_0.XCI" file, then change a minor parameter and click "OK". At this point, do not select "Out of context per IP", instead, choose "global". Then click "GENERATE". Next, click on the "PCIE_7X_0.XCI" file, select "Properties", find "IS_MANAGED" and turn it off. This action will make the files of the IP core editable. Expand the IP core folder, find the "pcie_7x_0_core_top.v" file, modify the aforementioned two parameters, and then save it. Afterwards, find and open the "bram_pcie_cfgspace.xci" file, click on "Other Options", edit your COE there and save it, and then click "OK", proceed to generate your bitstream file. By doing so, the configuration space should be as you desired. If not, try executing the reset_run synth_1 command and then regenerate the bitstream file.

zr5177, 2024-03-02:
After modifications like these, if the PC card likely fails during the boot phase, it could be due to the PCIe configuration space not adhering to specifications. Please cautiously handle every byte within the COE, and repeatedly experiment to ensure your parameter modifications occur at the intended location. Many tools that save configuration space files may have issues with endianness and formats not meeting expectations, requiring you to manually convert them to your desired format.

zr5177, 2024-03-02:
When the driver suspects that the device is not operating normally, it usually involves interactions based on the BAR (Base Address Register) space. It's not a problem that can simply be solved by emulating the configuration space. I think ekknod's pcileech-wifi could provide you with more inspiration.

kilmu-pro, 2024-03-02:
In the ekk project, it seems that he gives the return value of reading a specific address, rather than returning from bar.
So the value of the file bar zero4k.coe does not need to be modified?

zr5177, 2024-03-02:
I'm not familiar with the structure of the latest version of PCILEECH, as the BAR space implementation I'm currently using was built by myself before version 4.12. In my experience, most PCIe endpoint devices perform register mapping to the BAR in hardware, such as EEPROMs commonly used in wireless network cards. Therefore, you might want to learn ekknod's approach. As for more in-depth emulation, it might not be appropriate to discuss here.

ufrisk, 2024-03-02:
I'm closing this issue since it's not really an issue, more of a discussion. I think it will be possible to continue to post in the thread though, or keep the discussion going in some other forum.

Best wishes with your projects.

closed 2024-03-02 by ufrisk