ufrisk/pcileech-fpga #166: Leetdma Abnormal speed measurement

GRRENY, 2024-04-15:
I successfully produced leetdma using open source leetdma drawings, and successfully burned the squirrel's firmware, but something went wrong during the speed test. When the device is turned on, only the blue indicator flashes once, and the FT601 driver is successfully connected, but when you turn on the speed test to load, it will show that it is not connected to the FPGA device. I asked a lot of people, and they all said that the firmware did not match, but leetdma can burn the firmware of the squirrel normally. I was going crazy with this question, I didn't know what to do.
[image]
[image]
[image]

GRRENY, 2024-04-15:
Interestingly, I can't use the official burner provided by LeetDMA, but I successfully burned the firmware with the burner of HackDMA

ufrisk, 2024-04-15:
Nice to see you got it to work and congratulations on your new board.

But I'm closing this issue since I don't think successfully creating a board and flashing it is an issue.

closed 2024-04-15 by ufrisk