ufrisk/pcileech-fpga #168: Error when generating project files

VIPBotsMark, 2024-04-23:
Board: LightningZDMA 100T
OS: KUbuntu 23.04
Vivado: Vivado Lab Edition 2023.2

```bash
cd /home/mlx/Downloads/pcileech-fpga/ZDMA/

ls
100T
build.md
readme.md
vivado_build_100t.tcl
vivado_generate_project_100t.tcl

source vivado_generate_project_100t.tcl -notrace
ERROR: [Common 17-170] Unknown option '-part', please type 'create_project -help' for usage info.
```

The issue pops up when running from both the tcl shell environment, and from the tcl console in the Vivado GUI.
I've also tried win10 professional 22h2, same issue.

VIPBotsMark, 2024-04-23:
Output when running without notrace flag:  
```bash
source vivado_generate_project_100t.tcl
# set origin_dir "."
# if { [info exists ::origin_dir_loc] } {
#   set origin_dir $::origin_dir_loc
# }
# set _xil_proj_name_ "pcileech_tbx4_100t"
# if { [info exists ::user_project_name] } {
#   set _xil_proj_name_ $::user_project_name
# }
# variable script_file
# set script_file "vivado_generate_project.tcl"
# proc print_help {} {
#   variable script_file
#   puts "\nDescription:"
#   puts "Recreate a Vivado project from this script. The created project will be"
#   puts "functionally equivalent to the original project for which this script was"
#   puts "generated. The script contains commands for creating a project, filesets,"
#   puts "runs, adding/importing sources and setting properties on various objects.\n"
#   puts "Syntax:"
#   puts "$script_file"
#   puts "$script_file -tclargs \[--origin_dir <path>\]"
#   puts "$script_file -tclargs \[--project_name <name>\]"
#   puts "$script_file -tclargs \[--help\]\n"
#   puts "Usage:"
#   puts "Name                   Description"
#   puts "-------------------------------------------------------------------------"
#   puts "\[--origin_dir <path>\]  Determine source file paths wrt this path. Default"
#   puts "                       origin_dir path value is \".\", otherwise, the value"
#   puts "                       that was set with the \"-paths_relative_to\" switch"
#   puts "                       when this script was generated.\n"
#   puts "\[--project_name <name>\] Create project with the specified name. Default"
#   puts "                       name is the name of the project from where this"
#   puts "                       script was generated.\n"
#   puts "\[--help\]               Print help information for this script"
#   puts "-------------------------------------------------------------------------\n"
#   exit 0
# }
# if { $::argc > 0 } {
#   for {set i 0} {$i < $::argc} {incr i} {
#     set option [string trim [lindex $::argv $i]]
#     switch -regexp -- $option {
#       "--origin_dir"   { incr i; set origin_dir [lindex $::argv $i] }
#       "--project_name" { incr i; set _xil_proj_name_ [lindex $::argv $i] }
#       "--help"         { print_help }
#       default {
#         if { [regexp {^-} $option] } {
#           puts "ERROR: Unknown option '$option' specified, please type '$script_file -tclargs --help' for usage info.\n"
#           return 1
#         }
#       }
#     }
#   }
# }
# set orig_proj_dir "[file normalize "$origin_dir/../ScreamerM2_TEST4/pcileech_screamer_m2"]"
# set validate_required 0
# if { $validate_required } {
#   if { [checkRequiredFiles $origin_dir] } {
#     puts "Tcl file $script_file is valid. All files required for project creation is accesable. "
#   } else {
#     puts "Tcl file $script_file is not valid. Not all files required for project creation is accesable. "
#     return
#   }
# }
# create_project ${_xil_proj_name_} ./${_xil_proj_name_} -part xc7a100tfgg484-2
ERROR: [Common 17-170] Unknown option '-part', please type 'create_project -help' for usage info.
```

ufrisk, 2024-04-23:
I'm not able to replicate. It builds fine on Vivado 2023.2.2 for me on a freshly cloned project from Github.

[image]

VIPBotsMark, 2024-04-23:
Thank you for the fast reply.  
My guess is that the vivado version is the issue. 
Just grabbed the latest version from xilinx and it still persists. Tried cloning the project again just in case but no luck there.  
I will try to play around with the versions more, don't know what else it can be.

ufrisk, 2024-04-23:
Unfortunately I have no clue what the issue may be. I think others have used this project successfully already since there is some custom firmware for the ZDMA floating around out there.

Maybe someone in the PCILeech Discord have more ideas than I have

[image]

Since I don't believe this to be an issue with the actual project as such I'm closing this issue. But should you find out that there is please let me know so I may fix it.

closed 2024-04-23 by ufrisk

hajrs, 2024-04-28:
i think i have figured the issue, 

For anyone want to know what's the issue and how to fix 
 ERROR: [Common 17-170] Unknown option '-part', please type 'create_project -help' for usage info.  

The issue is with Vivado Lab edition, as it does not comes with the -part commands which what we saw in the TCL windows,

uninstalling the vivado lab edition and when installing do not choose the lab edition instead pick Vivado and then the full Vivado ML edition.

VIPBotsMark, 2024-04-28:
> i think i have figured the issue,
>
> For anyone want to know what's the issue and how to fix
>  ERROR: [Common 17-170] Unknown option '-part', please type 'create_project -help' for usage info.
>
> The [...]

Yup, just confirmed compiles just fine with ML Standard Edition. Thank you.  
@ufrisk please update the docs if you have time.