ufrisk/pcileech-fpga #172: hdl file for ZDMA's another fpga

w2016561536, 2024-06-27:
Hi, i want to ask will hdl file for ZDMA's another fpga be published in the future?

ufrisk, 2024-06-27:
It's quite unlikely as things look right now.

Please understand that when I created the PCILeech project and decided to do it as open source I never imagined that there would spawn a million dollar industry around it. The major players in this industry have seriously damaged this project by undercutting/outcompeting the only meaningful hardware sponsorship I had (LambdaConcept) whilst giving nothing back themselves.

This is perfectly fine since it's open source. And I'm very happy for my very kind Github Sponsors even though it in no way makes up for the loss/damage done to this project by the hardware vendors which contribute nothing to this project. Even so, I continue to put massive amounts of time into this project to keep it afloat. New features, bug fixes, and the upcoming support for Windows 11 24H2.

Publishing ZDMA FPGA1 as open source would be akin of shooting myself in the foot. It would completely destroy my remaining hardware sponsorships while the new hardware vendors using that design wouldn't contribute anything.

Also it would be nice to get something back financially for the many hours I put into making that HDL. Due to this it's highly unlikely I'd publish it as open source.

w2016561536, 2024-06-27:
Understand and respect to your decision. Have a nice day!

closed 2024-06-27 by w2016561536