ufrisk/pcileech-fpga #178: XC7A50T

VeelaCleave, 2024-09-22:
I have one of these lying around and wondering whether there would be any way to get it to work with PCILeech-FPGA.

I've been researching it for a few years on and off but never had the time to sit down properly.

I know PCILeech-FPGA would run on the chip if ported but is the FT2232 fast enough to interface if it's even possible to link it via a breakout board.

Schematic:

https://github.com/Surfndez/NanoEVB-X1/blob/master/EVB-block-diagram.png

ufrisk, 2024-09-22:
Everything will work given enough time put into it I guess, but would it be worth the time and effort?

You'd need to get some kind of carrier/breakout board to break out USB from the M2 connector while keeping the PCIe lanes directed towards the target computer.

Also using RS232 will be painfully slow, even tho it will work, and you'd have to adjust the pcileech firmware to support the RS232 transport, and also the LeechCore library communication with the FT2232 UART.

While at the same time you can get a [CaptainDMA M2 key M device for $170](https://captaindma.com/product/captain-dma-m-2/) or a PCIe board from LambdaConcept or CaptainDMA for around €150 - and everything would work out of the box - and really really fast as compared to running it over dog shit slow UART/RS232.

VeelaCleave, 2024-09-22:
> Everything will work given enough time put into it I guess, but would it be worth the time and effort?
>
> You'd need to get some kind of carrier/breakout board to break out USB from the M2 connector [...]

Thanks for the reply,

I've ordered a 75T Card as suggested!

Thanks for all your hard work!

ufrisk, 2024-09-22:
👍 And best wishes with your DMA project!

closed 2024-09-22 by ufrisk