ufrisk/pcileech-fpga #184: Cannot proceed Step 5 source vivado_build.tcl -notrace

lukrei, 2025-01-26:
Hello I tried to compile the firmware for the PCIeSquirrel Device. However on Step 5 i failed after a while with Error "A LUT1 cell in the design is missing a connection on input pin I0".

Expected Behaviour create "Xilinx proprietary IP cores and build bitstream".
Current behaviour: It fails with a error message instead see here:
Starting Connectivity Check Task
ERROR: [Opt 31-67] Problem: A LUT1 cell in the design is missing a connection on input pin I0, which is used by the LUT equation. This pin has either been left unconnected in the design or the connection was removed due to the trimming of unused logic. The LUT cell name is: i_pcileech_pcie_a7/i_pcie_7x_0/inst/inst/pcie_top_i/pcie_7x_i/pcie_block_i_i_1.
Resolution: Please review the preceding OPT INFO messages that detail what has been trimmed in the design to determine if the removal of unused logic is causing this error. If opt_design is being specified directly, it will need to be rerun with opt_design -verbose to generate detailed INFO messages about trimming.

Time (s): cpu = 00:00:00 ; elapsed = 00:00:00.002 . Memory (MB): peak = 1823.125 ; gain = 0.000
Phase 9.2 Verifying Netlist Connectivity | Checksum: 2457b0eee

Time (s): cpu = 00:00:07 ; elapsed = 00:00:06 . Memory (MB): peak = 1823.125 ; gain = 0.000
Phase 9 Finalization | Checksum: 2457b0eee

Time (s): cpu = 00:00:07 ; elapsed = 00:00:06 . Memory (MB): peak = 1823.125 ; gain = 0.000
INFO: [Common 17-83] Releasing license: Implementation
56 Infos, 22 Warnings, 0 Critical Warnings and 2 Errors encountered.
opt_design failed
ERROR: [Opt 31-67] Problem: A LUT1 cell in the design is missing a connection on input pin I0, which is used by the LUT equation. This pin has either been left unconnected in the design or the connection was removed due to the trimming of unused logic. The LUT cell name is: i_pcileech_pcie_a7/i_pcie_7x_0/inst/inst/pcie_top_i/pcie_7x_i/pcie_block_i_i_1.
Resolution: Please review the preceding OPT INFO messages that detail what has been trimmed in the design to determine if the removal of unused logic is causing this error. If opt_design is being specified directly, it will need to be rerun with opt_design -verbose to generate detailed INFO messages about trimming.
INFO: [Common 17-206] Exiting Vivado at Sun Jan 26 10:47:03 2025...
[Sun Jan 26 10:47:05 2025] impl_1 finished
ERROR: [Vivado 12-13638] Failed runs(s) : 'impl_1'
wait_on_runs: Time (s): cpu = 00:00:00 ; elapsed = 00:00:57 . Memory (MB): peak = 504.770 ; gain = 0.000
ERROR: [Common 17-39] 'wait_on_runs' failed due to earlier errors.

    while executing
"wait_on_run impl_1"
    (file "vivado_build.tcl" line 22)

ufrisk, 2025-01-26:
Are you building from the unmodified source code at: https://github.com/ufrisk/pcileech-fpga/tree/master/PCIeSquirrel or did you do any modifications at all, or trying to build some custom firmware?

is your source folder on disk directly under the root folder or in a relatively short path such as: C:\Temp\PCIeSquirrel   Building from a long path such as on your Desktop often causes problems.

lukrei, 2025-01-26:
yes i tried to change vendor id, device id and subsystem but at the moment i think i got the subsystem id in the wrong format since vivado wants it in 2 different fields.

lukrei, 2025-01-26:
After that process my DMA Card seems to malfunction. It is recognized as a INTEL Network Card after changing the 5 ID Initial Values under ID in vivado but it stops working. Then I tried to flash original firmware without my modifications.
I tried the following unmodified versions 4.13 and 4.14 and every time the firmware upgrade process seemed to successfully finish. Now the DMA Card is recognized as device id 666 again but the FTDI Port when changing to non firmware upgrade port seems missing in device manager. I restarted many times but it seems the DMA Card is broken since the FTDI USB Port in Device Manager does not come up again.

Can you help me what to do now?

ufrisk, 2025-01-26:
I'm sorry. I can't help with custom firmware. If it's not a bug in the project code as such the issues section here is not the place to report bugs in code you have written.

Please understand that this is an open source project, I don't get paid to do this.

Best wishes with your project though.

closed 2025-01-26 by ufrisk