ufrisk/pcileech-fpga #190: Yellow mark(Fw not working)

dnadma, 2025-04-17:
Captain 75t. Im have any problem with fw! In my device manager i have yellow mark and original wifi. I did all steps by one and customized it to my donor and in dma test its saying vmm init failed. How i can fix it

[image]

[image]

Noel18101987, 2025-04-20:
It's not a good idea to use an Ethernet device to develop firmware. Use another type of PCIe device and follow the guide at simonrak.se. Also, use the CaptainDMA/75t484 folder in the repository. To avoid problems, I recommend using Vivado 2023.2 and locating the project folder in C:/DMAProject, for example.

dnadma, 2025-04-20:
I already did any other PCIe devices same error...